Pi4J/pi4j-v1 #354: Controlling servo (modelcraft Y-3009) with pi4j and raspberry pi

framspott94, 2017-08-08:
Hello everyone,

I want to control a servo (modelcraft Y-3009) which is connected to the GPIO of my Raspberry Pi 3 and an separate voltage source. I have tried using the PWM function, but I cant create the desired signal for my servo. Also I tried to use the softPWM function, there I can get the desired timing for my servo, but the signal is distorted and it also does not work.

Has anybody a working example for this setup? Is this setup even possible, or do I have to use additional boards...

All the best - and thanks in advance!

framspott94

framspott94, 2017-08-10:
Hello,

After several other trials and another Pi, I finally got it working! Here is my resulting code, hopefully it is of use for another person...

`       com.pi4j.wiringpi.Gpio.wiringPiSetupGpio();
        
        int n = 18;
        System.out.println("Config Servo PWM with pin number: " + n);
        com.pi4j.wiringpi.Gpio.pinMode(n, com.pi4j.wiringpi.Gpio.PWM_OUTPUT);
        com.pi4j.wiringpi.Gpio.pwmSetMode(com.pi4j.wiringpi.Gpio.PWM_MODE_MS);
        com.pi4j.wiringpi.Gpio.pwmSetClock(192);
        com.pi4j.wiringpi.Gpio.pwmSetRange(2000);

        for(int i = 0; i < 5; i++){
            System.out.println("Set Servo");
            com.pi4j.wiringpi.Gpio.pwmWrite(n, 50);

            Thread.sleep(1000);

            System.out.println("Change servo state...");
            com.pi4j.wiringpi.Gpio.pwmWrite(n, 250);

            Thread.sleep(1000);

        }`

All the best

framspott94

salk31, 2017-09-08:
Excellent. I think you can close your own issue if you want to...