Pi4J/pi4j-v1 #363: Extend Servo abstractions to support turning off the servo (no pulses)

salk31, 2017-09-08:
ServoBlaster at least supports this but I can't see how the Servo interface exposes this ability.

I'm up for trying to create a patch if not a silly idea?

NB I'd be tempted to mark ValidationException as deprecated as only seems to be used be the servo code. I'll listen out for screams.

Cheers

Sam

natdan, 2017-09-08:
Great idea. Go for it! :)

salk31, 2017-09-09:
Leaning towards off() (copying the Power interface)... With the implementations I can't test just throwing UnsupportedOperationException until somebody wants to add an implementation.

salk31, 2017-09-09:
Using ServoBlaster I'm getting glitches. Maybe in some cases an already started pulse gets shortened?

closed 2018-05-02 by salk31