Pi4J/pi4j-v1 #364: Do I need to install pi4j manually if using a fat jar?

zharvey-welltok, 2017-09-08:
I'm looking at running a JVM app on a Raspberry Pi (**RPi**). To do this I need to:

1. Write the app
2. Build it & package it as a self-contained (shaded) executable JAR (that's the plan at least)
    * This means all of my lib dependencies get shaded into the jar and the whole bloated "fat jar" can be ran via `java -jar myapp.jar`
3. Install Raspbian on an SD card
4. Plug the SD card into the RPi, power it on, connect keyboard and monitor to it and manually install:
    * Java 8
    * Anything else I might need

For my app to communicate with the GPIO pins I plan on using the [pi4j library](http://pi4j.com/). According to the [installation docs](http://pi4j.com/install.html), pi4j depends on a Linux package called **WiringPi**. However, according to their docs, the WiringPi native library comes embedded in the pi4j jar:

> "*No installation is required for Wiring Pi. Wiring Pi is embedded in the Pi4J native library.*"

So my thinking is: I should be alright and my game plan (prescribed above) should work. My build system will package pi4j (which includes the embedded WiringPi native lib) and any other transitive dependencies into one giant fat jar and I should be able to run it.

**However**, their docs make **no mention of Maven/Gradle coordinates** for the pi4j jar. Instead, they recommend installing it manually on the RPi via:

    curl -s get.pi4j.com | sudo bash

This makes me think that perhaps something is wrong with my gameplan. So I ask:

* Is it OK that my app's only pi4j Gradle dependency is: `compile group: 'com.pi4j', name: 'pi4j-core', version: '1.1'`; and
* Do I still need to manually install anything on my RPi (see the curl command above)? Or will my gameplan (to build an executable fat jar) work?

Thanks in advance!

natdan, 2017-09-08:
Gradle dependency should be enough. You don't need to install pi4j on the target RPi.
. Mine (pom) has following:
```
        <dependency>
            <groupId>com.pi4j</groupId>
            <artifactId>pi4j-core</artifactId>
            <version>${pi4j.version}</version>
        </dependency>

        <dependency>
            <groupId>com.pi4j</groupId>
            <artifactId>pi4j-device</artifactId>
            <version>${pi4j.version}</version>
        </dependency>

        <dependency>
            <groupId>com.pi4j</groupId>
            <artifactId>pi4j-native</artifactId>
            <version>${pi4j.version}</version>
            <classifier>raspberrypi-dynamic</classifier>
            <type>so</type>
        </dependency>
```
I am not 100% sure if pi4j-native is actually needed, but that is quite easy to find out.

savageautomate, 2017-09-08:
> However, according to their docs, the WiringPi native library comes embedded in the pi4j jar:

Yes, Pi4J does depend on WiringPi but it is baked into (embedded in) the pi4j-core.jar file.  You don't have to install WiringPi separately. 

> Is it OK that my app's only pi4j Gradle dependency is: compile group: 'com.pi4j', name: 'pi4j-core', version: '1.1'; and

Yes; however, I would suggest using the 1.2-SNAPSHOT version at this time.  This version will be released very soon.

Here are instructions here on the Maven coordinates for the Pi4J project as well as details on the additional repository for using the SNAPSHOT builds:
http://pi4j.com/download.html

> Do I still need to manually install anything on my RPi (see the curl command above)? Or will my gameplan (to build an executable fat jar) work?

You don't have to install Pi4J using the curl command or the deb installer file.  You can simply package Pi4J jars with your project.  However, as for the "_fat_" JAR ...  that's where things can get a little complicated.  

Here is what's going on under the hood so you determine how best to package your app.  When Pi4J classes are loaded they need to resolve the pi4j native dependencies so they look for a native library named "libpi4j.so".  (_Example: https://github.com/Pi4J/pi4j/blob/master/pi4j-core/src/main/java/com/pi4j/wiringpi/Gpio.java#L189_)   There is a custom loader class (See _https://github.com/Pi4J/pi4j/blob/master/pi4j-core/src/main/java/com/pi4j/util/NativeLibraryLoader.java_) in Pi4J that looks for this library in the classpath.  As long as pi4j-core.jar is in the classpath, it is able to find the native library inside it, extract the appropriate "libpi4j.so" file for the given platform and use it for its runtime.  

I know this all works if you simply include the Pi4J-core.JAR along side your project JAR and make sure its loaded on the Java classpath when you start your app.  I have not tried and don't know what will happen in the case of a "fat" or "uber" JAR packaging.  It may all work just perfectly fine ... but if it complains at all about loading a native lib, then my guess is that NativeLibraryLoader can't find the native lib embedded in your JAR.  You may end up having to create a custom step in your build that forcefully includes the contents of the embedded pi4j-core.jar "lib" directory.    

Here you can see the Maven build step where we include the "lib" directory in the pi4j-core JAR:
https://github.com/Pi4J/pi4j/blob/master/pi4j-core/pom.xml#L280

So just make sure that your build does end up creating a "lib" directory with all the libpi4j.so content in the root path inside your final JAR.

Thanks , Robert

savageautomate, 2017-09-08:
@natdan 

> I am not 100% sure if pi4j-native is actually needed, but that is quite easy to find out.

I don't think the **pi4j-native** artifact is needed in your build.  The pi4j-native.jar is merely a container artifact for the compiled native libs and they get extracted from the pi4j-native.jar and copied directly into the pi4j-core.jar during the building of pi4j-core.  So its is needed for pi4j project compilations but should not be needed for third-party consumer projects.

closed 2017-09-08 by savageautomate

zharvey-welltok, 2017-09-11:
Thanks everyone -- I'm going to give it a try tonight and will report back if the fat jar method works. For the record, I'm using the [Gradle Shadowjar plugin](https://github.com/johnrengelman/shadow) which is super flexible and configurable.