Pi4J/pi4j-v1 #367: SerialDataEventListener stops receiving serial data

menazord, 2017-09-11:
Hi again!

I've found an issue with SerialDataEventListener... somehow, after 300+ iterations, the listener no longer receives data from serial port, even when serial connection was successfully established

This is my SerialDataEventListener

```
public class ArduinoDataListener implements SerialDataEventListener {

    private Logger LOGGER = LoggerFactory.getLogger(ArduinoDataListener.class);

    private Double temperature;

    private Long memory;

    @Override
    public void dataReceived(SerialDataEvent event) {

        String payload;

        try{
            payload = event.getAsciiString();
            LOGGER.debug("Payload received : " + payload);

            if (payload.startsWith(DeviceDataType.TEMPERATURE.getIdentifier())){
                payload = payload.replaceAll(DeviceDataType.TEMPERATURE.getIdentifier(), "");
                setTemperature(Double.parseDouble(payload));

            }else if (payload.startsWith(DeviceDataType.FREE_MEMORY.getIdentifier())){
                payload = payload.replaceAll(DeviceDataType.FREE_MEMORY.getIdentifier(), "");
                setMemory(Long.parseLong(payload));
            }

        }catch (IOException ioe){
            LOGGER.error("Error while processing serial payload", ioe);
        }
    }

    public Double getTemperature() {
        return temperature;
    }

    public void setTemperature(Double temperature) {
        this.temperature = temperature;
    }

    public Long getMemory() {
        return memory;
    }

    public void setMemory(Long memory) {
        this.memory = memory;
    }
}

```

And this is my service code (runs every X minutes via scheduling)

`
```
@Component("RaspberryPi")
public class RaspberryPi implements Device {

    private Logger LOGGER = LoggerFactory.getLogger(RaspberryPi.class);

    private static final Serial serial = SerialFactory.createInstance();

    private ArduinoDataListener arduinoDataListener;

[... 99 lines cut ...]

And this is my output for iteration 345

```
2017-09-11 03:31:00.001  INFO 1024 --- [pool-3-thread-1] c.f.gardenwatch.GardenwatchApplication   : Device RaspberryB sensing status for garden Piranha80
2017-09-11 03:31:00.513  INFO 1024 --- [pool-3-thread-1] c.f.g.s.sensing.SensingServiceImpl       : Starting sense cycle
2017-09-11 03:31:00.520  INFO 1024 --- [pool-3-thread-1] c.f.gardenwatch.device.pi.RaspberryPi    : Attempting connection to /dev/ttyACM0 , baud rate 9600
2017-09-11 03:31:00.581  INFO 1024 --- [pool-3-thread-1] c.f.gardenwatch.device.pi.RaspberryPi    : Connected
2017-09-11 03:31:00.587  INFO 1024 --- [pool-3-thread-1] c.f.gardenwatch.device.pi.RaspberryPi    : Querying sensor data..
2017-09-11 03:31:00.592 DEBUG 1024 --- [pool-3-thread-1] c.f.gardenwatch.device.pi.RaspberryPi    : Probing
2017-09-11 03:31:03.599 DEBUG 1024 --- [pool-3-thread-1] c.f.gardenwatch.device.pi.RaspberryPi    : Requesting TEMPERATURE
2017-09-11 03:31:09.613 DEBUG 1024 --- [pool-3-thread-1] c.f.gardenwatch.device.pi.RaspberryPi    : Requesting FREE MEMORY
2017-09-11 03:31:11.119  INFO 1024 --- [pool-3-thread-1] c.f.gardenwatch.device.pi.RaspberryPi    : Sensor data retrieved
2017-09-11 03:31:11.122 DEBUG 1024 --- [pool-3-thread-1] c.f.gardenwatch.device.pi.RaspberryPi    : Arduino memory status : 1587 free
2017-09-11 03:31:11.131  INFO 1024 --- [pool-3-thread-1] c.f.gardenwatch.device.pi.RaspberryPi    : Closing serial connection
2017-09-11 03:31:11.760  INFO 1024 --- [pool-3-thread-1] c.f.g.s.sensing.SensingServiceImpl       : Sensor data has been saved (id : 2645)
2017-09-11 03:31:11.768  INFO 1024 --- [pool-3-thread-1] c.f.g.s.sensing.SensingServiceImpl       : Sense cycle completed
2017-09-11 03:31:11.774  INFO 1024 --- [pool-3-thread-1] c.f.gardenwatch.GardenwatchApplication   : ========= STATUS ===========
2017-09-11 03:31:11.780  INFO 1024 --- [pool-3-thread-1] c.f.gardenwatch.GardenwatchApplication   :  Temperature   : 12.37 ºC
2017-09-11 03:31:11.799  INFO 1024 --- [pool-3-thread-1] c.f.gardenwatch.GardenwatchApplication   : ============================
```

Note that logging for "Payload received" no longer appears, and values for memory and temperature are the same as the last time they printed correctly (as values from listener do not reset)

Can you guide me to check if my code is using API correctly?

menazord, 2017-09-11:
Also seems related to thread "pi4j-single-executor-0"

menazord, 2017-09-12:
Hi

Executed another run last night, and data listener stopped working on iteration #355 .

It seems to always fail on this iteration, could this be a memory issue ?  I get no errors on logs, it just stops working

menazord, 2017-09-14:
Hi @savageautomate 

After testing several combinations and code enhancements, keeping the serial connection open during all program execution seems to solve the problem

I've achieved more than 3.000 iterations without any problems, and data is arriving consistently from remote serial devices.

savageautomate, 2017-09-15:
@menazord 

Thank you for your detailed thread and for narrowing down the problematic area to opening and closing the serial connection.  I'll have to have a look to see what all is going on under the covers during a serial closure.