Pi4J/pi4j-v1 #369: libpi4j.so and/or wiringPi are unable to determine hardware version

zharvey-welltok, 2017-09-15:
I am trying to run a Java 8 app on my **Raspberry Pi 1 Model A**, and use this library for driving the GPIO pins. I am using the [Gradle Shadow plugin](https://github.com/johnrengelman/shadow) to package my Java app up as a "*fat jar*"; that is, a self-contained executable jar (all transitive dependencies are shaded and merged into the jar; makes deploying JVM apps and their library dependencies much simpler).

I can confirm that after my build, my fat jar *does* contain a `libpi4j.so` file. When I decompile my jar I see it contains the following packages/contents:

[image]

However when I run my app on my RPi (via `java -jar starterbox.jar`), I get the following error message:

```
Unable to determine hardware version. I see: Hardware : BCM2835
,
- expecting BCM2708 or BCM2709.
If this is a genuine Raspberry Pi then please report this
to projects@drogon.net. If this is not a Raspberry Pi then you
are on your own as wiringPi is design to support the
Raspberry Pi ONLY.
```

**Anybody ever seen this before or know what the fix is?** According to [this post](https://www.raspberrypi.org/forums/viewtopic.php?t=182191), it seems that maybe updating the kernel might do the trick...but this is a brand-spanking-new Raspbian Stretch Lite image that I just got from [this link here](https://www.raspberrypi.org/downloads/raspbian/), so I'm not necessarily convinced I want to go that route.

Any ideas?

savageautomate, 2017-09-15:
Are you pulling in Pi4J 1.2-SNAPSHOT?  If you are using Pi4J version 1.1 or older, then yes its a known issue.

zharvey-welltok, 2017-09-15:
Ahhh OK thanks @savageautomate , I *did* try these coordinates:

```
<dependency>
    <groupId>com.pi4j</groupId>
    <artifactId>pi4j-core</artifactId>
    <version>1.1</version>
</dependency>
```

However when I try to resolve them (via `./gradlew eclipse` which pulls them from the remote repo and links them to my project's classpath inside Eclipse), I get errors:

    Could not resolve: com.pi4j:pi4j-core:1.2-SNAPSHOT

So I reverted the version to `1.1`.

I also don't see the snapshot jars in Maven Central...any ideas where I can find them?! Thanks so much!

savageautomate, 2017-09-15:
You will have to also add this custom repo to pull in snapshot builds.  Maven central does not allow snapshots.  

```
<repositories>
	<repository>
		<id>oss-snapshots-repo</id>
		<name>Sonatype OSS Maven Repository</name>
		<url>https://oss.sonatype.org/content/groups/public</url>
		<snapshots>
			<enabled>true</enabled>
			<updatePolicy>always</updatePolicy>
		</snapshots>
	</repository>
</repositories>

```
See:  http://pi4j.com/download.html#Maven_Repository

zharvey-welltok, 2017-09-15:
Thanks, I *think* that might have worked, but still seeing issues possibly relating to `libpi4j.so` and the fat jar build. Now when I run the jar I get the following error:

    java.lang.RuntimeException: Unable to open GPIO direction interface for pin[1]: Permission denied

Here's the code in my `main` class:

```
try {
    GpioController gpioController = GpioFactory.getInstance()
    GpioPinDigitalOutput myLed = gpioController.provisionDigitalOutputPin(RaspiPin.GPIO_01,
         'MyLED', PinState.HIGH)
} catch(Throwable t) {
    log.error(ExceptionUtils.getStackTrace(t))
}
```

Also I currently **dont** have anything hooked up to my GPIO pins (any of them), but I would think that would just cause the Java API to be inert; not throw a "Permission denied" error.

Any ideas on what's going on here? Thanks again!

savageautomate, 2017-09-15:
Are you running your Java program via "sudo"?  By default the Pi4J library needs administrative access to re-configure (_export_) GPIO pins on your program's behalf.  

There is another more recent option to avoid using "sudo" in some cases.   It may be a little out of date, but please see this article for more details and limitation on running without "sudo":
http://www.savagehomeautomation.com/pi4j-nonprivileged

savageautomate, 2017-09-15:
By the way .. the good news is that it looks like its working with your **fat** JAR.   It must be running some of the native code to get this far and realize there are GPIO permissions issues :-)

savageautomate, 2017-09-15:
You are welcome!

zharvey-welltok, 2017-09-15:
Also if you use Gradle and are curious how this fat jar config works, here's what I'm using:

```
import com.github.jengelman.gradle.plugins.shadow.transformers.ServiceFileTransformer

plugins {
    id 'maven-publish'
    id 'com.github.johnrengelman.shadow' version '1.2.3'
	id 'groovy'
	id 'eclipse'
	id 'application'
}

sourceCompatibility = 1.8
targetCompatibility = 1.8

group = 'mygroup'
mainClassName = 'com.me.MyApp'

repositories {
    jcenter()
    mavenCentral()
	maven {
		url 'https://oss.sonatype.org/content/groups/public'
	}
}

dependencies {
    compile(
      'org.codehaus.groovy:groovy-all:2.4.7'
      ,'org.slf4j:slf4j-api:1.7.25'
      ,'org.slf4j:slf4j-simple:1.7.25'
      ,'com.typesafe.akka:akka-actor_2.11:2.5.4'
	  ,'com.pi4j:pi4j-core:1.2-SNAPSHOT'
	  ,'org.apache.commons:commons-lang3:3.6'	  
    )

    testCompile 'org.spockframework:spock-core:1.0-groovy-2.4'
	testCompile 'junit:junit:4.12'
}

task wrapper(type: Wrapper) {
  gradleVersion = '3.5'
}

manifest {
	attributes 'Main-Class': mainClassName
}
jar {
	manifest {
		attributes 'Main-Class': mainClassName
	}
	baseName = 'myapp'
}

shadowJar {
	baseName = 'myapp'
	classifier = null
	version = null

	transform(ServiceFileTransformer) {
		exclude 'META-INF/*.DSA'
		exclude 'META-INF/*.RSA'
		exclude 'LICENSE*'
	}
	transform(com.github.jengelman.gradle.plugins.shadow.transformers.AppendingTransformer) {
		resource = 'reference.conf'
	}
	classifier = ''
}

publishing {
	publications {
		shadow(MavenPublication) {
			from components.shadow
			artifactId = 'myapp'
		}
	}
}
```