Pi4J/pi4j-v1 #375: Something wrong with PWM output

AlexanderShniperson, 2017-09-24:
Hi
i have a problem with pwm on motor driver, when i try to enable one of two motors there is runnig always two
my setup: raspberry pi 3 model b, motor driver controller L298N, jvm 8
my source [can be found here](https://github.com/AlexanderShniperson/pi4j_motordrivers])

Instruction:
- [Gradle build tool](https://gradle.org/releases/) needed to build a project
- to build project run command from project folder `gradle installDist`
- to run application `./build/TankClient/bin/TankClient`

need help to understand what i'm doing wrong.

AlexanderShniperson, 2017-09-25:
@savageautomate 
hi, can you please look at my problem?
thanks

eitch, 2017-09-25:
@AlexanderShniperson the link to your project is broken

AlexanderShniperson, 2017-09-25:
@eitch oh sorry, this is some typo in link, it is updated now

AlexanderShniperson, 2017-09-27:
looks like GPIO pins somehow related
GPIO 1  // supports PWM0
GPIO 26 // supports PWM0
GPIO 24 // supports PWM1

replaced second engine connection from GPIO_26 to GPIO_24 and now it works fine

fclinton, 2019-07-14:
Looks like all is good suggest closing @AlexanderShniperson

closed 2019-07-18 by AlexanderShniperson