Pi4J/pi4j-v1 #378: Insecure easy install method

mattconsto, 2017-10-18:
Currently the recommended, easy way for a user to install Pi4J is as follows:

    curl -s get.pi4j.com | sudo bash

This is terrible for many reasons, the worst is that it doesn't use HTTPS and it cannot use HTTPS as get.pi4j.com doesn't respond to HTTP requests. As it is completely insecure, an attacker could potentially MitM the connection, or the user's connection could simply be flaky. I haven't checked the downloaded code, but if you are not writing checks to ensure the script has fully downloaded, the users machine could execute a partially downloaded script.

My recommendation is to enable HTTPS on get.pi4js.com and change the documentation to download the script over HTTPS. 

More info: https://stackoverflow.com/a/34016579/3990396

savageautomate, 2017-10-18:
You make an excellent point.   I'll look into adding support for HTTPS.  Its hitting an Amazon S3 server in the background so hopefully I can somehow associate a SSL cert with the S3 bucket.

ryankenney, 2018-09-15:
I juuust about tossed this entire project out as unusable because there was not way to reliably establish trust for installation, then I found the github sources. Maybe just tie to github instead of pi4j.com?

savageautomate, 2019-03-07:
As of version 1.2 release, the Pi4J website, download links as well as the install/uninstall scripts and API repositories are all now secured using HTTPS/SSL.

* [Pi4J Website](https://www.pi4j.com)
* [Pi4J Downloads](https://www.pi4j.com/1.2/download.html)
* [Pi4J Installation](https://www.pi4j.com/1.2/install.html)

Here is the updated simple install command:
```
curl -sSL https://pi4j.com/install | sudo bash
```
However, please note that as always, this is not the only method to install.  You can also download the *.deb package file and install it manually, download the raw compiled JARs or download and compile the raw source code.

closed 2019-03-07 by savageautomate

mattconsto, 2019-03-19:
Thanks a lot @savageautomate for fixing this!