Pi4J/pi4j-v1 #387: How to read string from Arduino via I2C

Tonyboy78, 2017-11-17:
Dear members!

I want to read a string from Arduino and I am confused about how to do that. In my code below I am receiving just the first letter of the word "Hello". I read many examples but I don't understand them because I am new to Java programming. So if there is someone who can help me to fix that ai will be very grateful. Here is my code so far:
```
import java.io.IOException;
import com.pi4j.io.i2c.I2CBus;
import com.pi4j.io.i2c.I2CDevice;
import com.pi4j.io.i2c.I2CFactory;
public class I2C_Test2 {
	public static void main(String[] args) throws Exception {
		try {
	            System.out.println("Creating I2C bus");
	            I2CBus bus = I2CFactory.getInstance(I2CBus.BUS_1);
	            
	            System.out.println("Create I2C device 1");
	            I2CDevice device = bus.getDevice(0x8);
                       String data="";
			           for( int i = 0 ; i < 5 ; i++ ){
				      data = data+ (char)device.read();
				    }
				    System.out.println(data);
                      } catch (IOException ex) {
			  ex.printStackTrace(); }

```

avivfox93, 2018-07-15:
Hello, this should do the work:

```
import java.io.IOException;
import com.pi4j.io.i2c.I2CBus;
import com.pi4j.io.i2c.I2CDevice;
import com.pi4j.io.i2c.I2CFactory;
public class I2C_Test2 {
	public static void main(String[] args) throws Exception {
		try {
	            System.out.println("Creating I2C bus");
	            I2CBus bus = I2CFactory.getInstance(I2CBus.BUS_1);
	            
	            System.out.println("Create I2C device 1");
	            I2CDevice device = bus.getDevice(0x8);

                    byte[] buffer= new byte[5];
                    device.read(buffer,0,buffer.length);
                    String data = new String(buffer);

                    System.out.println(data);
                      } catch (IOException ex) {
			  ex.printStackTrace(); }
```
in order to read more then one byte at once, you need to use:
read(byte[] buffer, int offset, int size);

savageautomate, 2021-02-15:
No further activity.  Closed.

For reference, here is example code of Pi4J communicating to an Arduino board via SPI.  (The Arduino requires slave SPI code and must be 3VDC or level shifted.)
https://github.com/Pi4J/pi4j/tree/master/pi4j-core/src/test/java/com/pi4j/io/hardware/i2c

closed 2021-02-15 by savageautomate