Pi4J/pi4j-v1 #388: How to read an Array from Arduino via I2C and Java

Tonyboy78, 2017-11-21:
Dear members!
I want to read an Array from Arduino and I am confused about how to do that. In my code below I am receiving just the first number of the integers {1,2,3,4}. I read many examples but I don't understand them because I am new to Java programming. So if there is someone who can help me to fix that ai will be very grateful. Here is my code so far:
```
import java.io.IOException;
import com.pi4j.io.i2c.I2CBus;
import com.pi4j.io.i2c.I2CDevice;
import com.pi4j.io.i2c.I2CFactory;
public class I2C_Test2 {
    public static void main(String[] args) throws Exception {
        try {
                System.out.println("Creating I2C bus");
                I2CBus bus = I2CFactory.getInstance(I2CBus.BUS_1);

                System.out.println("Create I2C device 1");
                I2CDevice device = bus.getDevice(0x8);
                    int []f =new int[5];

                    for( int i = 0 ; i < f.length ; i++ )
                    {
                       f[i]=device.read();
                        }
                  System.out.println(Arrays.toString(f));
                  } catch (IOException ex) {
              ex.printStackTrace(); }
```
Here is the "Slave" code - Arduino:
```
#include <Wire.h>
#define SLAVE_ADDRESS 0x5 // Slave board with adress 0x5
#define SIZE 5

void setup() {
  // initialize i2c as slave
  Wire.begin(SLAVE_ADDRESS);

  // define callbacks for i2c communication
  Wire.onRequest(sendData);
  Serial.begin(9600);
}

void loop() {
}

void sendData() {
byte buf[] = {1, 2, 3, 4};
Wire.write (buf, sizeof (buf));
}
```

natdan, 2017-11-22:
I think you got i2c bus wrong: you cannot read byte per byte of the bus. It would, then, read just first byte from device again and again. What you need is reading an array of bytes in one go. Try
```
int read(byte[] buffer, int offset, int size) throws IOException;
```
method (from here http://pi4j.com/apidocs/com/pi4j/io/i2c/I2CDevice.html). Something like:
```
int readLegth = device.read(f, 0, f.length);
```

Tonyboy78, 2017-11-23:
Thank you for the answer! Now I am receiving: 1,2,3,4 with this code:
```
 byte[] f = new byte[4];
 device.read(f, 0, f.length);
      for( int i = 0 ; i < f.length ; i++ ) {
	System.out.println(f[i]);
     }
```