Pi4J/pi4j-v1 #391: Proper synchronous GPIO pin pulse methods

NogginBops, 2017-11-24:
Currently there is no proper way to call the `pulse()` method synchronously. There is a flag in [GpioPinImpl.java - pulse()](https://github.com/Pi4J/pi4j/blob/master/pi4j-core/src/main/java/com/pi4j/io/gpio/impl/GpioPinImpl.java) to say that the call should block. 
```java
pulse(long duration, PinState pulseState, boolean blocking, Callable<Void> callback)
```

But this generates a problem when the thread gets interrupted.
It throws a `RuntimeException` instead of a `InterruptedException`.
Because the methods normal use-case does not throw an exception it would be inconvenient to add a `throws` declaration to the method. 
But this makes a lot of sense when you want to call it synchronously where you expect to get `InterruptedException`s from blocking calls.
Shouldn't there be synchronous methods available that throw the appropriate exception?

savageautomate, 2021-01-14:
*Housekeeping*

Pi4J v2 includes separate synchronous and asynchronous pulse methods.

closed 2021-01-14 by savageautomate

NogginBops, 2021-01-14:
This is still a problem.
It's still impossible to call `pulse( blocking = true )` because of the `RuntimeException`.
https://github.com/Pi4J/pi4j/blob/master/pi4j-core/src/main/java/com/pi4j/io/gpio/impl/GpioPinImpl.java#L351

This should be changed to an `InterruptedException`.

savageautomate, 2021-01-14:
**Fixed for v1.3.**  Added a compliment of `pulseSync()` methods to digital pin outputs.  Deprecated `pulse()` methods that contain the `blocking` parameter.  This way we don't break the interface for existing implementations.  

https://github.com/Pi4J/pi4j/blob/b9630767db601e74ef94534017862f0f13688199/pi4j-core/src/main/java/com/pi4j/io/gpio/GpioPinDigitalOutput.java#L73-L76

commit b0042d61 (2021-01-14): Fixed #391; Deprecated GPIO pulse() methods that included optional 'blocking' parameter and added pulseSync() methods which throw InterruptedException;