Pi4J/pi4j-v1 #392: pi4j项目只能操作针脚吗？我不懂呀，哪里有使用实例或者api说明

hftang, 2017-11-30:


natdan, 2017-11-30:
Try here: http://pi4j.com/
Or documentation:  http://pi4j.com/apidocs/index.html
Also, examples are here:
http://pi4j.com/example/control.html
http://pi4j.com/example/listener.html
http://pi4j.com/example/shutdown.html
http://pi4j.com/example/trigger.html
http://pi4j.com/example/more.html

Also, if you look at the code you can find a few useful tips in here for instance:
https://github.com/Pi4J/pi4j/tree/master/pi4j-example/src/main/java

HTH

ShisanXing, 2019-12-19:
这个项目是不是已经停止维护了，作者全身心投入pi4j-v2，新特性支持JDK11，但会不会放弃对jdk8的支持了？而且把树莓派4b放在v2里支持，这个要等好久的呀！！

savageautomate, 2021-01-16:
**Closed.**

New feature development has moved to Pi4J v2.x.  Version 1.x is still being maintained for any major issues.  However, the WiringPi library used in Pi4J V1.x has been DEPRECATED so we will encourage users to migrate to Pi4J v2.x when it gets released.  In the meantime, v1.3 (JDK8) and v1.4 (JDK9) are planned for release very soon.  Both of these will support the latest RPI devices including Model 4B, 400 and CM4.

closed 2021-01-16 by savageautomate