Pi4J/pi4j-v1 #397: How can I count the pluses generated by hardware PWM exactly ?

nanjsun, 2017-12-16:
I want to use Raspberry Pi 3 B to control a step motor exactly, and I want to count the number of pluses generated by PWM  exactly, I want to know:
1. Is there a method to count PWM pluses?
2. If there is no such a method , How can I do this by a timer interrupt in accurate of microsecond(or higher), or is there a timer in such accurate ?

Thanks!

rlsutton1, 2017-12-16:
It will not be possible to count pwm pulses accurately in java. While you could write the code to do it sooner or later a garbage collection (GC) will cause you to to miss some events.

I had written code to read quadrature encoding java, but found it to be very inaccurate due to GC causing events to be missed. I resorted to using some C to monitor the quadrature.

Also driving a stepper dircectly out of java is problematic for the same reasom. So long as you don't try to drive it so fast that a an unexpected pause in the stepping will cause it to lose position you will be ok. I have a 200 step per turn stepper, I cant reliably spin it any faster than about 1 turn per second.

NB: the minimum sleep period in java is 1 ms, so i use busy waiting and System.getNanoTime to achieve microsecond pauses, but there will always be GC pauses of arbitary amounts of time (many milliseconds )

nanjsun, 2017-12-17:
@rlsutton1 
Thanks for your reply, and your answer helps me a lot.

closed 2018-08-27 by savageautomate