Pi4J/pi4j-v1 #398: SPI read sometimes shifts bytes, sometimes bits

mukarl, 2017-12-18:
I'm using pi4j out of a multitasking environment and read from AD/DA board from waveshare (https://www.waveshare.com/wiki/High-Precision_AD/DA_Board), which has http://www.ti.com/lit/ds/symlink and /ads1256.pdf http://www.ti.com/lit/ds/symlink/dac8532.pdf on board.

with significant:-( probability I get conversion values, which are eight shifted by a byte, so can be perfectly corrected by division by 256. sometimes only one bit is missing, giving values twice too high.

is there any synchronisation problem known?

furthermore, for very small values, i get wrong encoded negativ values, where one 0xFF byte is coming as 0x00.

I use simple synchronized prefix on my high level methods, should be sufficient in order not to interact with other tasks, isn't it?

any problem known?

regards
Karl

mukarl, 2017-12-18:
forgot to mention, on a RPI 3b, with jessie light,
and http://www.ti.com/lit/ds/symlink/ads1256.pdf is the correct link

cljk, 2018-09-29:
Possible that you just had a race condition with your threads?

I dontt believe accessing the SPI is thread safe

savageautomate, 2021-02-15:
No further activity.  Closed.

closed 2021-02-15 by savageautomate