Pi4J/pi4j-v1 #404: pi4j doesn't compile with java 9

bsutton, 2018-03-01:
I've just upgraded to java 9 and when trying to build pi4j got the following errors:

pi4j-core/src/main/java/com/pi4j/io/file/LinuxFile.java:[34,16] cannot find symbol
symbol:   class Cleaner
location: package sun.misc
pi4j-core/src/main/java/com/pi4j/io/file/LinuxFile.java:[35,16] cannot find symbol
symbol:   class SharedSecrets
location: package sun.misc
pi4j/pi4j-core/src/main/java/com/pi4j/io/file/LinuxFile.java:[215,32] cannot find symbol
symbol:   variable SharedSecrets
location: class com.pi4j.io.file.LinuxFile
pi4j-core/src/main/java/com/pi4j/io/file/LinuxFile.java:[313,15] cannot find symbol
symbol:   class Cleaner
location: class com.pi4j.io.file.LinuxFile

albahrani, 2018-03-01:
With JDK9 access to the sun-packages has been restricted by default.
You may use the _-jdkinternals_ flag when compiling to get it working without change right now.
On the long run the following should be done:

- [ ] Replace uses of `sun.misc.Cleaner` with `java.lang.ref.Cleaner` in _LinuxFile.java_
- [ ] Replace uses of `sun.misc.SharedSecrets`with `jdk.internal.misc.SharedSecrets` in _LinuxFile.java_

TangPere, 2018-07-22:
@albahrani There should be jdk.internal.ref.Cleaner instead of java.lang.ref.Cleaner

agilkes, 2018-08-04:
In LinuxFile.java, I have replaced the "sun.misc" imports with imports of jdk.internal.misc.SharedSecrets and jdk.internal.ref.Cleaner. But when I "build" the pi4j.core project in Eclipse, the Java compiler (10.0.2) gives me "is not accessible" error messages for both imports. These result in 3 other compiler error messages that appear later in LinuxFile.java. How do you get the Java compile to succeed?

TangPere, 2018-08-05:
@agilkes  You should add compiler parameters "--add-exports=java.base/jdk.internal.ref=ALL-UNNAMED --add-exports=java.base/jdk.internal.misc=ALL-UNNAMED"

closed 2021-02-15 by savageautomate