Pi4J/pi4j-v1 #408: pi4j throws exception on Raspberry Pi 3 B+

sflebrun, 2018-04-08:
Has anyone tried to use pi4j on the new Raspberry Pi 3 B+?

The RPi 3B+ requires the use of wiringPi version 2.46, released sometime after March 14, 2018 and the latest version of Raspbian "stretch".

The pi4j jar files include the library libpi4j.so which is the embedded copy of wiringPi.

When I try to run the statement:
`
protected GpioController  gpio = GpioFactory.getInstance();
`

I get the following execption:

OpenJDK 64-Bit Server VM warning: You have loaded library /tmp/libpi4j6824788762589179225.so which might have disabled stack guard. The VM will try to fix the stack guard now.
It's highly recommended that you fix the library with 'execstack -c <libfile>', or link it with '-z noexecstack'.
Apr 07, 2018 9:26:14 PM com.pi4j.util.NativeLibraryLoader load
**SEVERE: Unable to load [libpi4j.so] using path: [/lib/raspberrypi/static/libpi4j.so]**
java.lang.UnsatisfiedLinkError: /tmp/libpi4j6824788762589179225.so: **/tmp/libpi4j6824788762589179225.so: wrong ELF class: ELFCLASS32 (Possible cause: architecture word width mismatch)**
	at java.lang.ClassLoader$NativeLibrary.load(Native Method)
	at java.lang.ClassLoader.loadLibrary0(ClassLoader.java:1941)
	at java.lang.ClassLoader.loadLibrary(ClassLoader.java:1824)
	at java.lang.Runtime.load0(Runtime.java:809)
	at java.lang.System.load(System.java:1086)
	at com.pi4j.util.NativeLibraryLoader.loadLibraryFromClasspath(NativeLibraryLoader.java:148)
	at com.pi4j.util.NativeLibraryLoader.load(NativeLibraryLoader.java:97)
	at com.pi4j.wiringpi.Gpio.<clinit>(Gpio.java:189)
	at com.pi4j.io.gpio.RaspiGpioProvider.<init>(RaspiGpioProvider.java:69)
	at com.pi4j.io.gpio.RaspiGpioProvider.<init>(RaspiGpioProvider.java:51)
	at com.pi4j.platform.Platform.getGpioProvider(Platform.java:125)
	at com.pi4j.platform.Platform.getGpioProvider(Platform.java:118)
	at com.pi4j.io.gpio.GpioFactory.getDefaultProvider(GpioFactory.java:109)
	at com.pi4j.io.gpio.impl.GpioControllerImpl.<init>(GpioControllerImpl.java:56)
	at com.pi4j.io.gpio.GpioFactory.getInstance(GpioFactory.java:91)
	at com.lebruns.steven.epaper.WaveshareEP_SPI.<init>(WaveshareEP_SPI.java:110)
	at com.lebruns.steven.epaper.WaveshareEPDisplay.<init>(WaveshareEPDisplay.java:20)
	at com.lebruns.steven.epaper.Waveshare27BFrame.<init>(Waveshare27BFrame.java:23)
	at com.lebruns.steven.epaper.epaper.main(epaper.java:29)

Exception in thread "main" java.lang.UnsatisfiedLinkError: com.pi4j.wiringpi.Gpio.wiringPiSetup()I
	at com.pi4j.wiringpi.Gpio.wiringPiSetup(Native Method)
	at com.pi4j.io.gpio.RaspiGpioProvider.<init>(RaspiGpioProvider.java:69)
	at com.pi4j.io.gpio.RaspiGpioProvider.<init>(RaspiGpioProvider.java:51)
	at com.pi4j.platform.Platform.getGpioProvider(Platform.java:125)
	at com.pi4j.platform.Platform.getGpioProvider(Platform.java:118)
	at com.pi4j.io.gpio.GpioFactory.getDefaultProvider(GpioFactory.java:109)
	at com.pi4j.io.gpio.impl.GpioControllerImpl.<init>(GpioControllerImpl.java:56)
	at com.pi4j.io.gpio.GpioFactory.getInstance(GpioFactory.java:91)
	at com.lebruns.steven.epaper.WaveshareEP_SPI.<init>(WaveshareEP_SPI.java:110)
	at com.lebruns.steven.epaper.WaveshareEPDisplay.<init>(WaveshareEPDisplay.java:20)
	at com.lebruns.steven.epaper.Waveshare27BFrame.<init>(Waveshare27BFrame.java:23)
	at com.lebruns.steven.epaper.epaper.main(epaper.java:29)
`
Is this problem due to using an out of date version of wiringPi?  If so, are there any plans for spinning a new version of pi4j with the wiringPi v2.46?

Is there any way to force pi4j to use an external version of wiringPi?

Jon-McEwan, 2018-04-13:
I had a similar issue an i solved it by compiling wiringPi v2.46 for Pi4J myself using the existing maven structure, and then replacing the libraries in my apps final jar file.

To compile wiringPi v2.46 for Pi4J:

1. clone the pi4j repo to your pc (this one https://github.com/Pi4J/pi4j)

2. change the used wiring-pi repo to the offical one (this is currrently v2.46):
open pi4j/pi4j-native/src/main/native/build-raspberrypi.sh and change **WIRINGPI_REPO** to git://git.drogon.net/wiringPi 
(ie uncomment line 83: _**export WIRINGPI_REPO=git://git.drogon.net/wiringPi**_
and comment out line 84: _**#export WIRINGPI_REPO=https://github.com/Pi4J/wiringPi)**_

3. remove the static argument from make command (as of v2.46 wiring-pi no longer supports staticaly linking see https://git.drogon.net/?p=wiringPi;a=blob;f=wiringPi/noMoreStatic;h=1d259420cd57dee992d6a8bda9ac3543ba8eff0a;hb=HEAD):
open pi4j/pi4j-native/src/main/native/wiringpi-build.sh and make two changes:
- change line 84 from _**make clean all static -j1 $@**_ to _**make clean all -j1 $@**_
- comment out line 85 _**mv libwiringPi.a ../lib/static/libwiringPi.a**_

4. change parent pom properties:
- lines 111-115 need to be changed to the correct values for you pi ie:
        <pi4j.default.host>**_your pi's hostname here**_</pi4j.default.host>
        <pi4j.default.port>22</pi4j.default.port>
        <pi4j.default.user>pi</pi4j.default.user>
        <pi4j.default.password>**_your password here_**</pi4j.default.password>
        <pi4j.default.directory>pi4j-temp</pi4j.default.directory>

- enable raspberry pi build: change line 126 to _**<raspberrypi.build>true</raspberrypi.build>**_

5. navigate to pi4j/pi4j-native and run **mvn package**

6. the libraries are now in **pi4j\pi4j-native\target\native\lib\raspberrypi**

Just replace the libraries in your jar file with these and it will work.

I haven't tested it extensively but I haven't found any issues yet.

savageautomate, 2018-04-23:
I was able to update the Pi4J 1.2-SNAPSHOT build today with the latest WiringPi.  I have not had a change to thoroughly test it but it is available if you want to try it again.

**Please Note:**  I have removed the default Pi4J native library static compile against WiringPi.  Now  Pi4J only dynamically links against WiringPi and thus you must have WiringPi installed on your target Raspberry Pi system.  (WiringPi is already included by default in the latest Raspbian images)

sflebrun, 2018-05-01:
The original problem that I posted was not caused by an out of date WiringPi library running on a Raspberry Pi 3 B+.   I would have hit that problem if I had gotten past this one.  This problem appears to be a problem with remote debugging with Eclipse.  The Java application runs fine on the Raspberry Pi 3 B+, with the updated Pi4J library that links dynamically to the WiringPi library and crashes when it is run in remote debug mode through Eclipse.

I tested my guess on a working Java Application on a Raspberry Pi 3 B.  It also crashes when run in remote debugging mode from Eclipse where the Eclipse is running on a Linux Mint system.  This application was successful run in this mode.  The source code was not recompiled or relinked since its last successful debugging session.  The only change that I am aware of is that Eclipse has been updated.

More work is needed to identify that actual cause.

sflebrun, 2018-05-01:
Problem solved.  **It was a case of pilot error**.  I configured debug in Eclipse to be local debug instead of remote debug.  So Eclipse was looking for an ELFCLASS32 library when it was finding one compiled to run on the ARM chip in the Raspberry Pi.

savageautomate, 2018-05-06:
This issue should be resolved with the latest 1.2-SNAPSHOT builds. 
Please reopen if there is still any issue.

closed 2018-05-06 by savageautomate