Pi4J/pi4j-v1 #409: Unable to determine hardware for model 3 B plus (BCM2835)

matthewleonard, 2018-04-16:
I received this error when trying to rovision a digital output pin: 

`Unable to determine hardware version. I see: Hardware	: BCM2835
,
 - expecting BCM2708 or BCM2709.
If this is a genuine Raspberry Pi then please report this
to projects@drogon.net. If this is not a Raspberry Pi then you
are on your own as wiringPi is designed to support the
Raspberry Pi ONLY.
`

I am using version 1.1 of pi4j
`        <dependency>
            <groupId>com.pi4j</groupId>
            <artifactId>pi4j-core</artifactId>
            <version>1.1</version>
        </dependency>
`

This is the code that I am trying to execute

`    public boolean setValue(int equipmentId, String value) {
        System.out.println("Setting "+equipmentId+" value of "+value);
        PinState state = value=="0"?PinState.LOW:PinState.HIGH;
        GpioPinDigitalOutput equipment = gpio.provisionDigitalOutputPin(RaspiPin.getPinByAddress(equipmentId),
                "Equipment",
                PinState.LOW);
        equipment.setState(state);
        return true;
    }`

Debian Version: `9.4`

CPU info:
`processor	: 0
model name	: ARMv7 Processor rev 4 (v7l)
BogoMIPS	: 38.40
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd03
CPU revision	: 4

processor	: 1
model name	: ARMv7 Processor rev 4 (v7l)
BogoMIPS	: 38.40
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd03
CPU revision	: 4

processor	: 2
model name	: ARMv7 Processor rev 4 (v7l)
BogoMIPS	: 38.40
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd03
CPU revision	: 4

processor	: 3
model name	: ARMv7 Processor rev 4 (v7l)
BogoMIPS	: 38.40
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd03
CPU revision	: 4

Hardware	: BCM2835
Revision	: a020d3
Serial		: 000000005f55642c
`

savageautomate, 2018-04-23:
I was able to update the Pi4J 1.2-SNAPSHOT build today with the latest WiringPi.  I have not had a change to thoroughly test it but it is available if you want to try it again.

m00fighter, 2018-04-25:
I was experiencing the same effect, updated my libraries to the snapshot build, and for me, all is OK. Thank you! :) (I hope it's OK for others too.)

matthewleonard, 2018-04-25:
I switched to the 1.2 snapshot and all has been working well so far. Thanks for the information!

I saw the note that we should now ensure that wiringPi is installed on the pi that runs the Pi4j application so I’m good there as well

closed 2018-04-25 by matthewleonard

SandroMoretti, 2020-12-17:
> 
> 
> @m00fighter
> 
> Thanks for the update and glad to hear its working for you now.
> 
> **As a reminder to others, you must have WiringPi installed on your system with the latest 1.2-SNAPSHOT builds.**
> 
> ```
> sudo apt-get install wiringPi
> ```
> 
> Thanks, Robert

wiringPi with "P" upper case got me error.
If someone is trying install this in future, just do with "p" low case.

```
sudo apt-get install wiringPi
```