Pi4J/pi4j-v1 #410: I'm using 64-bit Ubuntu on my pi, but pi4j is 32-bit

Xsxdot, 2018-04-24:
SEVERE: Unable to load [libpi4j.so] using path: [/lib/raspberrypi/dynamic/libpi4j.so]
java.lang.UnsatisfiedLinkError: /tmp/libpi4j7854506006999061126.so: /tmp/libpi4j7854506006999061126.so: wrong ELF class: ELFCLASS32 (Possible cause: architecture word width mismatch)

Can I get 64-bit pi4j?

sorry my chinese english

Xsxdot, 2018-04-29:
OK,I'll try it. 

Thanks.

closed 2018-08-27 by savageautomate

jannemec, 2020-01-29:
Hi did you succeed? I think more and more 64bit installations will be used.

Xsxdot, 2020-01-29:
Yes,but I deleted this version Ubuntu.
I'm sorry!

发自我的iPhone

megahertzon, 2020-07-01:
Hi friends, when I'll try to compile on 64 bit ubuntu system, a error appear:

[INFO] ------------------------------------------------------------------------
[INFO] Reactor Summary for Pi4J :: Parent POM 1.3-SNAPSHOT:
[INFO] 
[INFO] Pi4J :: Parent POM ................................. SUCCESS [ 14.897 s]
[INFO] Pi4J :: JNI Native Library ......................... FAILURE [ 30.239 s]
[INFO] Pi4J :: Java Library (Core) ........................ SKIPPED
[INFO] Pi4J :: GPIO Extension ............................. SKIPPED
[INFO] Pi4J :: Device Abstractions ........................ SKIPPED
[INFO] Pi4J :: Java Examples .............................. SKIPPED
[INFO] Pi4J :: Distribution ............................... SKIPPED
[INFO] ------------------------------------------------------------------------
[INFO] BUILD FAILURE
[INFO] ------------------------------------------------------------------------
[INFO] Total time:  58.488 s
[INFO] Finished at: 2020-06-30T22:19:37-05:00
[INFO] ------------------------------------------------------------------------
[ERROR] Failed to execute goal org.apache.maven.plugins:maven-antrun-plugin:1.8:run (pi4j-build-native-libraries) on project pi4j-native: An Ant BuildException has occured: The following error occurred while executing this line:
[ERROR] /home/citofonia/pi4j/pi4j-native/build.xml:304: The following error occurred while executing this line:
[ERROR] /home/citofonia/pi4j/pi4j-native/build.xml:210: The following error occurred while executing this line:
[ERROR] /home/citofonia/pi4j/pi4j-native/build.xml:85: exec returned: 127
[ERROR] around Ant part ...<ant inheritRefs="true" target="build-libpi4j.so-local">... @ 4:59 in /home/citofonia/pi4j/pi4j-native/target/antrun/build-pi4j-build-platform-libraries.xml
[ERROR] -> [Help 1]
org.apache.maven.lifecycle.LifecycleExecutionException: Failed to execute goal org.apache.maven.plugins:maven-antrun-plugin:1.8:run (pi4j-build-native-libraries) on project pi4j-native: An Ant BuildException has occured: The following error occurred while executing this line:
/home/citofonia/pi4j/pi4j-native/build.xml:304: The following error occurred while executing this line:
/home/citofonia/pi4j/pi4j-native/build.xml:210: The following error occurred while executing this line:
/home/citofonia/pi4j/pi4j-native/build.xml:85: exec returned: 127
around Ant part ...<ant inheritRefs="true" target="build-libpi4j.so-local">... @ 4:59 in /home/citofonia/pi4j/pi4j-native/target/antrun/build-pi4j-build-platform-libraries.xml

how I can fix them??
Best regards,

Nelson

Xsxdot, 2020-07-01:
@jannemec I can't see anything from the problem you described. But I found a version I compiled earlier. It might be useful to you.
@megahertzon Do you still need?
Where should I upload?

Xsxdot, 2020-07-01:
Backup is really the best thing in this world!

megahertzon, 2020-07-01:
hi friend can you share me your working version please?

Xsxdot, 2020-07-01:
sudo apt-get update
sudo apt-get upgrade
sudo apt-get -y install build-essential
sudo apt-get -y install cmake git libgtk2.0-dev pkg-config libavcodec-dev libavformat-dev libswscale-dev
sudo apt-get -y install python-dev python-numpy libtbb2 libtbb-dev libjpeg-dev libpng-dev libtiff-dev libjasper-dev libdc1394-22-dev
sudo apt-get -y install gcc g++
sudo apt-get -y install less
sudo apt-get -y install make
sudo apt-get -y install ant
sudo apt-get -y install tree
tar zxf 64-pi.tar.gz
tar zxf jdk-arm64.tar.gz
sudo mv jdk1.8.0_171/ /opt/
sudo mv apache-maven-3.5.3 /opt/

[pi4j-1.2-SNAPSHOT.deb.zip](https://github.com/Pi4J/pi4j/files/4855547/pi4j-1.2-SNAPSHOT.deb.zip)

This is my init.sh at the time.
But I also encountered other problems at that time, I have forgotten how to solve it

FDelporte, 2020-07-01:
FYI @savageautomate has done a lot of work the last days on building a 32 and 64 bit version with Docker for V1.4 and V2. An ongoing process, but you can already find more info on https://v2.pi4j.com/build.

megahertzon, 2020-07-01:
Great man Xsxdot, your deb package seems to work pi4j on 64bit orange pi SBC

Xsxdot, 2020-07-01:
Oh yeah!

发自我的iPhone

------------------ Original ------------------
From: megahertzon <notifications@github.com&gt;
Date: Wed,Jul 1,2020 9:11 PM
To: Pi4J/pi4j <pi4j@noreply.github.com&gt;
Cc: Xsxdot <10811836@qq.com&gt;, Author <author@noreply.github.com&gt;
Subject: Re: [Pi4J/pi4j] I'm using 64-bit Ubuntu on my pi, but pi4j is 32-bit (#410)

savageautomate, 2020-07-01:
The `develop/1.4` branch includes binaries for both 32-bit and 64-bit Raspberry Pi. 
The `develop/1.4` branch also includes updated to builder scripts to compile both 32-bit and 64-bit native libraries either via a Docker container or using ARM cross-compilers.  

See https://github.com/Pi4J/pi4j/pull/494