Pi4J/pi4j-v1 #411: Unable to load [libpi4j.so] using path: [/lib/raspberrypi/dynamic/libpi4j.so]

chhaysereiseth, 2018-04-24:
**Dear 
I use Pi 3 B, I had run my code with pi4j for along but I got this error for today. I think I missing something.**

Apr 24, 2018 5:52:56 PM com.pi4j.util.NativeLibraryLoader load
SEVERE: Unable to load [libpi4j.so] using path: [/lib/raspberrypi/dynamic/libpi4j.so]
java.lang.UnsatisfiedLinkError: /tmp/libpi4j5334134479954495835.so: libwiringPi.so: cannot open shared object file: No such file or directory
        at java.lang.ClassLoader$NativeLibrary.load(Native Method)
        at java.lang.ClassLoader.loadLibrary0(ClassLoader.java:1938)
        at java.lang.ClassLoader.loadLibrary(ClassLoader.java:1821)
        at java.lang.Runtime.load0(Runtime.java:809)
        at java.lang.System.load(System.java:1086)
        at com.pi4j.util.NativeLibraryLoader.loadLibraryFromClasspath(NativeLibraryLoader.java:159)
        at com.pi4j.util.NativeLibraryLoader.load(NativeLibraryLoader.java:105)
        at com.pi4j.wiringpi.Gpio.<clinit>(Gpio.java:189)
        at com.pi4j.io.gpio.RaspiGpioProvider.<init>(RaspiGpioProvider.java:69)
        at com.pi4j.io.gpio.RaspiGpioProvider.<init>(RaspiGpioProvider.java:51)
        at com.pi4j.platform.Platform.getGpioProvider(Platform.java:125)
        at com.pi4j.platform.Platform.getGpioProvider(Platform.java:118)
        at com.pi4j.io.gpio.GpioFactory.getDefaultProvider(GpioFactory.java:109)
        at com.pi4j.io.gpio.impl.GpioControllerImpl.<init>(GpioControllerImpl.java:53)
        at com.pi4j.io.gpio.GpioFactory.getInstance(GpioFactory.java:91)
        at com.khmerwikigroup.rgbled.RGBLed.<init>(RGBLed.java:32)
        at com.khmerwikigroup.AppLoader.<clinit>(AppLoader.java:27)

savageautomate, 2018-04-25:
@chhaysereiseth 

Starting on 2018-04-23, the 1.2-SNAPSHOT builds no longer statically links to the WiringPi dependency and you must have WiringPi installed on your system separately from Pi4J.

```
sudo apt-get install wiringPi
```

Or build and install from source:
http://wiringpi.com/download-and-install/

This is part of the effort to support the new RPi3B+ and in accordance with the changes requested by WiringPi.

chhaysereiseth, 2018-04-25:
Thank you so much.

closed 2018-04-25 by chhaysereiseth