Pi4J/pi4j-v1 #417: I see: Hardware BCM2835

sinkingtheyap, 2018-06-15:
This is still happening with the following setup:

**GPIO Version:** gpio version: 2.46
**Raspberry Pi Details:**
  Type: Pi 2, Revision: 01, Memory: 1024MB, Maker: Sony 
  * Device tree is enabled.
  *--> Raspberry Pi 2 Model B Rev 1.1
  * This Raspberry Pi supports user-level GPIO access.

Pi4J Version: 
--------------------------------------------
            THE Pi4J PROJECT
--------------------------------------------
   VERSION   : 1.2-SNAPSHOT
   TIMESTAMP : 2018-04-23 04:22:34
--------------------------------------------

The error is: 
<--Pi4J--> GPIO Debounce Example ... started.
Unable to determine hardware version. I see: Hardware	: BCM2835
,
 - expecting BCM2708 or BCM2709.
If this is a genuine Raspberry Pi then please report this
to projects@drogon.net. If this is not a Raspberry Pi then you
are on your own as wiringPi is designed to support the
Raspberry Pi ONLY.

sinkingtheyap, 2018-06-15:
SOLUTION: sudo rpi-update 52241088c1da59a359110d39c1875cda56496764
This is the only thing that worked. A Firmware change.

savageautomate, 2018-08-27:
Pi4J now depends on the WiringPi that you have installed on your system.  So you probably had a newer version of WiringPi that was expecting the latest RPI firmware.

closed 2018-08-27 by savageautomate