Pi4J/pi4j-v1 #418: New 1.2 release?

bsutton, 2018-06-19:
Could we get a new 1.2 release so we have an up to date build available in maven etc?

ak1959, 2018-08-11:
Agreed - current snapshot is as good or better than 1.1

savageautomate, 2018-08-27:
I agree and will try to get this out in the next couple of weeks.  I'm just so far behind in other projects and parts of my life that this project has been on the back burner for a while now.

eitch, 2018-08-27:
@savageautomate If you want, i can help with making new versions. I am an open source developer myself ([Strolch](https://github.com/4treesCH/strolch)) and thus know what it means to get new versions out there.

camueller, 2018-09-02:
**There is really a need for a new release!** I was on 1.2-SNAPSHOT to get support for a new board and suddenly wiringpi was required to be installed. It is really dangerous to use a snapshot that long since everytime I build (and release!) my own application I might get a new snaphort of pi4j with a different behaviour.

eitch, 2018-11-16:
@savageautomate It has been quite some time now, is it possible that you could find the time to make the new release? Or perhaps i could help you there?

thegreystone, 2018-11-23:
Is there an ETA? I would be happy to help.

mirage22, 2018-11-23:
I'd be happy to help too

natdan, 2018-11-23:
Or just ask Robert Savage for dev rights and do it here? :)

eitch, 2018-11-23:
Well, i did say i would like to help and thought that implied dev rights.

So here: @savageautomate

Would you give me dev rights to help with making a new release?

johanjanssen, 2018-12-19:
@savageautomate please let @eitch or someone else help you to create a release. Pi4J is a great and very popular solution. However people are desperate for a new release. I for one don't like to use snapshot releases. It would be great if this project is continued because else there's basically no other option then creating a fork. 

Please don't take this personal, I appreciate the hard and good work you've done and I understand that at the moment you have other priorities, but please let people help you :).

savageautomate, 2019-02-26:
@eitch 

Sorry for the delay ... I have been pretty ill since the start of this year and unfortunately have not been following this project and its tickets closely.

If you are still interested, I would be happy to add you to the developers list/permissions.

Thanks, Robert

savageautomate, 2019-02-26:
@johanjanssen 

I'm working on a 1.2 release right now.    I know its way overdue.   I was hoping to tackle a number of open issues and get them all included in 1.2, but I think I just need to get 1.2 released and work on any further fixes in a new future build.  

Thanks, Robert

johanjanssen, 2019-02-26:
@savageautomate thanks a lot for working on the new release and adding more contributors. 

Hope you're feeling better now!

eitch, 2019-02-26:
@savageautomate Hey Robert, sorry to hear you were sick and hope you are doing better now.

Yes, i am still interested in helping out with releasing. So if you want to add me that would be fine. I might need a few pointers in building the release. But i am sure that won't be too hard, as i have my own project Strolch release to Maven as well.

savageautomate, 2019-02-26:
@eitch 

I'm going to go ahead and try to get v1.2 out the door pretty much as-is.  
We can start any new efforts/bug fixes in v1.3.

I was also thinking about moving to a v2 where we might want to significantly reduce some of the project bloat.  I think it may have been too ambitious to include everything that Pi4J is currently attmepting to tackle.   At least from a support perspective, I would rather reduce the complexity and re-focus on only the main objectives which were to simply support the RAW I/O for the Raspberry Pi.  

On the list to potentially cut would be support for other platforms (outside of RaspberryPi).  The WiringPi support for other boards simply is not very well maintained and it causes a lot of regression testing and incompatibilities.   I would also consider removing support for GPIO extenders (maybe not the idea/interfaces, but the concrete impls) and third-party platforms/boards.  I would probably also deprecate the "pi4j-device" project that attempted to focus on device and component level APIs.  Those never really were fully developed for the original goals intended.

Thoughts? 

Thanks, Robert

camueller, 2019-02-26:
> I was also thinking about moving to a v2 where we might want to significantly reduce some of the project bloat. I think it may have been too ambitious to include everything that Pi4J is currently [...]
I fully agree with that which is also underlined by the project name. I'm looking forward to see a 1.2 release :-)

Axel

thegreystone, 2019-02-26:
Any thinking around support for JDK 11? There are now (at least) two nice OpenJDK 11 based distributions - the Azul one and the Bell-Soft one.

eitch, 2019-02-27:
@savageautomate 

It definitely makes sense to just release 1.2 as is and start a new 1.3 or even 2.x. I myself use 1.2 snapshot in production as 1.1 doesn't work for me anymore and would really prefer a release instead of a snapshot. This is probably what the rest of the community would prefer too.

For bugfixes etc. we can then still to 1.2.x or even 1.3.x releases if need by and then do the major changes in 2.x, i.e. master/develop branch.

johanjanssen, 2019-02-27:
> 
> 
> Any thinking around support for JDK 11? There are now (at least) two nice OpenJDK 11 based distributions - the Azul one and the Bell-Soft one.

I would love to have Java 11 support as well. Maybe as a next step after a as is 1.2 release.

Reducing the project also sounds like a good idea to make it easier to maintain and upgrade.

mirage22, 2019-02-27:
I've compiled pi4j 1.2 SNAPSHOT on java11 but, I've not tested it yet. I've used a travis setup.

savageautomate, 2019-02-27:
JDK 11 will be targeted in the v2.0 builds. (coming soon)

thegreystone, 2019-02-27:
See #442.

savageautomate, 2019-02-27:
The current `master` branch is prepped for a 1.2 release.  Feel free to clone and compile it on your own if you want to check/test anything before 1.2 gets released.

Thanks. Robert

eitch, 2019-02-27:
Splendid. I'll do a test build when i am at home, as i don't have a raspi with me at the moment.

eitch, 2019-02-28:
I did a small test and don't see any problems. I think releasing should be fine.

eitch, 2019-02-28:
Btw, i only compiled for the raspberry pi, as i don't have any of the other devices.

savageautomate, 2019-02-28:
@eitch 

No worries, thanks for testing it out.

savageautomate, 2019-02-28:
## Pi4J v1.2 is now released.
https://github.com/Pi4J/pi4j/releases/tag/release%2F1.2

It may take a day or two for it to get synchronized into the Maven Central repo.

thegreystone, 2019-02-28:
Congrats! :)

eitch, 2019-02-28:
Splendid!

savageautomate, 2019-02-28:
Website now updated too :-)

savageautomate, 2019-02-28:
And .... sync with Maven Central now complete ...
https://search.maven.org/search?q=g:com.pi4j

_That took less time than I expected._

closed 2019-02-28 by savageautomate

mirage22, 2019-03-05:
Congratulation ! 
I've compiled pi4j for JDK 11 and tested with Bellsoft liberica JDK 11.0.2 looks like all is running.  
simple test: LCD over i2c with keyboard

savageautomate, 2019-03-07:
@mirage22 
I'm working on the v2.0 source branch and trying to get it compiling for JDK11.

What did you do to get the project compiling with JDK 11 with regards to `LinuxFile.java` and the incompatible imports:
```
import sun.misc.Cleaner;
import sun.misc.SharedSecrets;
``` 

Thanks, Robert

savageautomate, 2019-03-07:
@mirage22 

The `develop/2.0` branch now compiles against JDK 11 as well.
https://github.com/Pi4J/pi4j/tree/develop/2.0

mirage22, 2019-03-07:
Great! 
@savageautomate I'll test it and give a feedback!