Pi4J/pi4j-v1 #425: Problem with 1.1

CharlesGodwin, 2018-08-26:
I have a problem with the version 1.1

I was using pi4j-1.0-SNAPSHOT for years to read a USB/RS485 device. It's a Magnum Energy system that sends packets of bytes and the packets size is identified by a pause in the data stream. Using the older version, I was able to read a character at a time and the delay between packets was reliably measurable.

I have just upgraded to 1.1 and the old read() method to return a character is gone. I have attempted other routines but I'm not getting any measurable delay between characters. I suspect that the software is buffering everything nd the delays are being washed out.

The old, working, code is available here [https://goo.gl/i6VQSp](url)

I don't know of any other way to read this data as it's poorly structure and delay between packets, which provides message length, was the only reliable method to decipher the messages.

Or... i'm missing something obvious in the protocol or use of the Pi4j API.

Any suggestions anyone can offer would be appreciated. 

This link will provide a description of the methodology used.

[http://static.us-inverters.com/docs/support/Magnum-Networking-Communications-Protocol-%282009-10-15%29.pdf](url)

savageautomate, 2018-08-27:
Hi Charles,

Yes, in 1.1 the serial interfaces were changed to provide better support for a standardized interface which includes buffering, eventing, etc.    

- Added new Serial JNI implementation to support more advanced serial configuration/operations
- Updated Serial interface to extend from AutoCloseable.
- Fixed #135; Serial communication using 7 Data Bits not working.
- Fixed the Pi4J Serial interface to use the new Serial JNI implementation
- Fixed wiringPi serial JNI methods to better support raw data operations

You could try the older WiringPi interfaces (which are still included) to access the serial port and data -- they still include a method to return a single byte and should perform similar to the older Pi4J serial impl.  

SEE:
- https://github.com/Pi4J/pi4j/blob/master/pi4j-core/src/main/java/com/pi4j/wiringpi/Serial.java
- https://github.com/Pi4J/pi4j/blob/master/pi4j-example/src/main/java/WiringPiSerialExample.java

You may also want to upgrade to version 1.2-SNAPSHOT if you need compatibility with the latest RPI hardware boards.

Thanks, Robert

CharlesGodwin, 2018-08-27:
I tried your suggestion. I have pi4j 1.1 installed on my Rpi 

I have two essentially identical programs except for the call to pi4j.
Here is the invocations
the original
` java -classpath magnum.jar:lib/* Test_rs485 records=10`
The revised
` java -classpath godwin.jar:/opt/pi4j/lib/'*':lib/'*' Test_rs485a records=10`

The first one behaves as expected and has been for a few years. The original is using pi4j jar files in my folder and the revised is using the ones provided by the pi4j Easy install.

I have attached the two files. What am I doing wrong?
[pi4j-issue425.zip](https://github.com/Pi4J/pi4j/files/2325109/pi4j-issue425.zip)

savageautomate, 2018-08-27:
I'm not 100% sure what going wrong, but you may need to make a call to:

```Serial.serialDataAvail(fd)```
https://github.com/Pi4J/pi4j/blob/master/pi4j-core/src/main/java/com/pi4j/wiringpi/Serial.java#L229

before attempting to read the byte in and verify that a byte is available to be read.

---

I also looked up the old Pi4J 1.0 implementation and its serial impl did in fact use the WiringPi serial APIs under the hood. 

SEE:  
https://github.com/Pi4J/pi4j/blob/release/1.0/pi4j-core/src/main/java/com/pi4j/io/serial/impl/SerialImpl.java#L139

So the old `serial.read()` method was just calling the `com.pi4j.wiringpi.Serial.serialGetchar(fd)` under the hood.

```
    /**
     * <p> This method will read the next character available from the serial port receive buffer.</p>
     * <p>
     * <b>NOTE: If a serial data listener has been implemented and registered with this class, then
     * this method should not be called directly. A background thread will be running to collect
     * received data from the serial port receive buffer and the received data will be available on
     * via the event.</b>
     * </p>
     * 
     * @return next available character in the serial data buffer
     */
    public char read() throws IllegalStateException {
        
        // validate state
        if (isClosed()) 
            throw new IllegalStateException("Serial connection is not open; cannot 'read()'.");

        // attempt read byte from serial port 
        return (char) com.pi4j.wiringpi.Serial.serialGetchar(fileDescriptor);
    }
```

---

Another approach you could try is just to read the available bytes in the buffer and measure the time in between the counts.  When you see your expected delay time, then you could read in the expected/available number of bytes from the buffer at one time.  That might be slightly more efficient.  It sucks that you only have a pause delay as a delimiter between packets.

savageautomate, 2018-08-27:
You could also try making your own custom Serial class by copying the old impl class from:
https://github.com/Pi4J/pi4j/blob/release/1.0/pi4j-core/src/main/java/com/pi4j/io/serial/impl/SerialImpl.java
and removing all the monitoring stuff since you don't need it.  

It's also possible that something changed in the WiringPi library related to its internal Serial impl.  I'm not sure - that would take some investigation into the WiringPi change history.

CharlesGodwin, 2018-08-28:
Thanks for your help. I re-evaluated my approach and now have  a working solution.

closed 2018-08-28 by CharlesGodwin