Pi4J/pi4j-v1 #427: What pins the /dev/ttyS0 is mapped to after the changes mentioned below?

HitChand, 2018-09-01:
Hello,

I want to connect my custom board having RS232 interface to Raspberry Pi 3 Model B. 

It is mentioned on the page [](http://pi4j.com/pins/model-3b-rev1.html) :-

> On the Raspberry Pi Model 3B the hardware-based serial/UART device /dev/ttyAMA0 has been re-purposed to communicate with the the built-in Bluetooth modem and is no longer mapped to the serial RX/TX pins on the GPIO header. Instead, a new serial port "/dev/ttyS0" has been provided which is implemented with a software-based UART (miniUART). This software-based UART ("/dev/ttyS0") does not support PARITY and some have experienced some stability issues using this port at higher speeds. If you don't need Bluetooth functionality, you can disable the BT modem and configure the RPi to use a device-tree overlay to re-map the hardware-based serial UART ("/dev/ttyAMA0") back to the GPIO header pins for TX/RX. 

**I understand that /dev/ttyS0 can be used to have RS 232 connections; but to which pins of the J8 Pinout (40-pin Header) should I connect my custom boards Rx Tx so that they can communicate using RS232 protocol?**

Thanks.

HitChand, 2018-09-01:
http://pi4j.com/pins/model-3b-rev1.html

savageautomate, 2021-01-11:
**Closed.** 
Pins 15 & 16.

closed 2021-01-11 by savageautomate