Pi4J/pi4j-v1 #429: hardware version error with latest NOOBs on RaspberryPi 3B+

shishirtab, 2018-10-04:
I referred http://pi4j.com/example/control.html and used simple code to send high signal to a pin. Getting following error during run although it compiled fine.
javac -classpath .:classes:/opt/pi4j/lib/'*' -d . ControlGpioExample.java

sudo java -classpath .:classes:/opt/pi4j/lib/'*' ControlGpioExample

Error:
Unable to determine hardware version. I see: Hardware	: BCM2835
,
 - expecting BCM2708 or BCM2709.

I checked GPIO version. It is 2.46
WiringPi is also latest one, still facing this issue with Raspberry Pi 3 B+ 
-- 
uname -a
Linux raspberrypi 4.14.50-v7+ #1122 SMP Tue Jun 19 12:26:26 BST 2018 armv7l GNU/Linux

haggy, 2019-02-22:
What is the estimated release date for the snapshot containing this fix?

shishirtab, 2019-02-22:
It is already available.

savageautomate, 2019-02-26:
@haggy 

I'm working on a release build of v1.2 now.  Look for it in a few days.

closed 2019-02-26 by savageautomate