Pi4J/pi4j-v1 #432: GPIO pin 16 (BCM) breaking serial

eudyptula, 2018-11-19:
If I provision RaspiBcmPin.GPIO_16 as a digital output pin, I can no longer receive any data on the serial port. It remains broken until I reboot the Pi, just stopping the application doesn't fix it. Provisioning any of the pins 12, 24, and 25 works fine - only seen it with pin 16 so far.

I suspect this may be an Pi4J issues since the equivalent Python code works fine with no problems at all, but of course could also be some configuration difference between the two libraries that I missed.

Pi 3 Model B, Raspbian 8, Pi4J v1.1 from Maven

DanielMartensson, 2018-11-21:
Can you post error code?  

Try Pi4J V1.2 instead.

eudyptula, 2018-11-27:
@DanielMartensson No error code, just get nothing from serial if I do:
```
greenLED = gpio.provisionDigitalOutputPin(RaspiBcmPin.GPIO_16, PinState.HIGH);
```

If I do the same with another Pin it works fine, e.g.:
```
redLED = gpio.provisionDigitalOutputPin(RaspiBcmPin.GPIO_12, PinState.HIGH);
```

Switching 1.2-SNAPSHOT from the Sonatype OSS Maven Repository, gives me (when running it on the Pi):
```
Caused by: java.lang.ClassNotFoundException: com.pi4j.io.gpio.GpioProvider
        at java.net.URLClassLoader.findClass(URLClassLoader.java:381)
        at java.lang.ClassLoader.loadClass(ClassLoader.java:424)
        at sun.misc.Launcher$AppClassLoader.loadClass(Launcher.java:331)
        at java.lang.ClassLoader.loadClass(ClassLoader.java:357)
        ... 3 more
```

DanielMartensson, 2018-11-27:
What Pi do you use? Remember that all pins are not digital outputs.

Use the .jar file instead from Pi4J zip file. Maven can be buggy.

savageautomate, 2021-01-14:
_I realize this is an old ticket, but I'm reviewing issues for the upcoming v1.3 release._

This issue sounds very suspicious.  

Provisioning `RaspiBcmPin.GPIO_16` should export BCM pin number [36](https://pinout.xyz/pinout/pin36_gpio16) on the 40 pin header.  However, it sounds like something is getting applied to pin number [10](https://pinout.xyz/pinout/pin10_gpio15) on the 40 pin header which is the RXD line on the default UART/serial port.  Coincidentally,  `RaspiPin.GPIO_16` is assigned to pin number [10](https://pinout.xyz/pinout/pin10_gpio15 on the 40 pin header.  So it sure seems like some piece of code is blindly evaluating the number 16 and doing something to the UART RX pin and not honoring the difference between the BCM and WiringPi numbering scheme.

savageautomate, 2021-01-14:
I just tested this on the latest Pi4J v1.3-SNAPSHOT build on a RPi 4B with latest *unofficial* WiringPi v2.60 (see: 
 https://pi4j.com/1.3/install.html#WiringPi_Native_Library) ...

Everything seemed to work fine.  No errors and it did not lock up the serial port which in my case is connected to the RPi's shell/console.  

It could have been an older WiringPi issue ... or it definitely could have been if you were missing the following line before provisioning the pins.

```
// in order to use the Broadcom GPIO pin numbering scheme, we need to configure the
// GPIO factory to use a custom configured Raspberry Pi GPIO provider
GpioFactory.setDefaultProvider(new RaspiGpioProvider(RaspiPinNumberingScheme.BROADCOM_PIN_NUMBERING));
```  

Closing ticket as tested and not reproducible with latest Pi4J version.

closed 2021-01-14 by savageautomate