Pi4J/pi4j-v1 #438: ArrayIndex Out of Bounds in case the adress is 100

JozefDropco, 2019-01-14:
Hi, 
this is a great library to use with Raspberry. I really appreciate your effort. 

There is a small bug in code which causes array out of bounds. While checking the Pin cache size you are comparing adress with array length. Instead it should be length-1 as the index cannot be 100 this caused an issue in my code.
java.lang.ArrayIndexOutOfBoundsException: 100
	at com.pi4j.io.gpio.GpioProviderBase.getPinCache(GpioProviderBase.java:88)
	at com.pi4j.io.gpio.GpioProviderBase.export(GpioProviderBase.java:119)
	at com.pi4j.io.gpio.GpioProviderBase.export(GpioProviderBase.java:100)
	at com.pi4j.io.gpio.impl.GpioPinImpl.export(GpioPinImpl.java:158)

I would like also to ask is there any plan for release 1.2?

savageautomate, 2019-02-26:
FIX #438; array auto-expansion should be triggered if requested pin address is GREATER THAN or EQUAL to the current pin cache length.

savageautomate, 2019-02-26:
Yes, working on a 1.2 release now.

commit c240b733 (2019-02-26): FIX #438; array auto-expansion should be triggered if requested pin address is GREATER THAN or EQUAL to the current pin cache length.
closed 2019-02-26 by savageautomate