Pi4J/pi4j-v1 #440: benchmark

miathedev, 2019-01-19:
Actually, it would be interesting for some people how performant with which VM pi4j is.
So i though, lets benchmark it.

I tested the pi zero with the standard openjdk-8-headless package. I got ~28-29kHz with the following code:

```
// create gpio controller
final GpioController gpio = GpioFactory.getInstance();

// provision gpio pin #01 as an output pin and turn on
final GpioPinDigitalOutput pin = gpio.provisionDigitalOutputPin(RaspiPin.GPIO_03, "MyLED",PinState.HIGH);
        
pin.setShutdownOptions(true, PinState.LOW);
        
while(true) {
  pin.toggle();
}
```

Can people that have something else than a pi zero benchmark it?
It would be really interessting. I would like to create a comparison table.

haggy, 2019-03-07:
@miathedev The tool I call [Pins-N-NeedlesI](https://github.com/haggy/pins-n-needles) that I currently made available on my GH, while definitely not a benchmark tool, has shown me how performant Pi4J and java IO with raspberry pi can be. Im able to program and switch circuits involving simple IC's like shift registers __very__ fast. You might find it interesting to play around with while you analyze the performance!

savageautomate, 2021-01-11:
**Closed**.  Lack of further activity.

closed 2021-01-11 by savageautomate