Pi4J/pi4j-v1 #442: Make Pi4J JDK 11 compatible

thegreystone, 2019-02-19:
Either branch a new major version of Pi4J that only supports JDK 11+, or reflectively access things like sun.misc.SharedSecrets vs. jdk.internal.misc.SharedSecrets.

I'd be happy to help.

savageautomate, 2019-02-27:
Agreed.  This will be targeted in version 2.0.  We will start a new branch for 2.0 soon.

thegreystone, 2019-02-27:
Very good. The biggest decision will be whether or not to go all-in JDK 11 or not. By all-in I mean modules, and using the JDK 11 APIs exclusively (no reflective access, tunneling or proxying to bind to the right version of a sun.misc class etc). I just started a branch in the Robo4J fork where I’m planning on going all in. That is the thing Miro was referring to in the other issue.

savageautomate, 2019-02-27:
I would prefer to get rid of any reflective access, etc. moving forward.  If you want to help with this effort, I can add you to the developers list.

thegreystone, 2019-02-27:
Sure, would be happy to help. I am currently designing a version 2 of Coff-E (my robot), so will have to get this up and running anyways (Robo4J now require JDK Flight Recorder, open sourced in JDK 11). When are you planning on setting up the 2.0 branch?

savageautomate, 2019-02-27:
Soon.  Working to get 1.2 out the door first.  So probably next week.

savageautomate, 2019-02-27:
I also want to see about changing the open source license for the project in V2.  We had to comply with the GPL license in previous versions and force the project to inherit the GPL license because we were compiling/linking in the WiringPi project.  As of 1.2 we are not longer statically linking against WiringPi and in V2 we will drop support for the other platforms that may still have static linking against WiringPi platform derivatives.

thegreystone, 2019-02-27:
Excellent! I will be traveling the next two weeks, but will bring a pi with me to have something to do in the weekends. I’ve ordered an ydlidar G4 and some other fun stuff that I’ll pick up when in the US. :)

savageautomate, 2019-03-04:
**2.0 branch created:**
https://github.com/Pi4J/pi4j/tree/develop/2.0

Untested at this point, but I have started removing parts of the project we no longer intend on maintaining/supporting.

thegreystone, 2019-03-05:
Excellent! Do you want me to work directly in the branch, or do you want me to fork and do PRs?

savageautomate, 2019-03-05:
@thegreystone 

What do you think is best/most efficient?   I'm open to working directly in the branch until we get a 2.0 release.  That way there is not a single gatekeeper.  However, we probably need a means for collaborative communication on v2.0 project efforts.  Any suggestions?  I was thinking of maybe using a private forum, but maybe some project management app like Basecamp would be better?

thegreystone, 2019-03-05:
We're using Slack in some other projects, and it seems to be working fairly well. Problem with slack is that the history is limited, and the illusion of persistence is just that - an illusion. OpenTracing is using Gitter. Not sure what the Gitter policy is in terms of persistence. Never tried Basecamp myself. Mailinglists are used as the primary channel for "offical" communication in OpenJDK, and IRC for more informal discussions. For JMC there is a slack workspace as well.

thegreystone, 2019-03-06:
Is there, by any chance, a Pi4J slack available? Otherwise I could set one up, and we could start there and see how it works for us?

savageautomate, 2019-03-07:
@thegreystone 

I have actually already started setting up a forum.  I wanted a tool that can handle a bit more organizational structure and support and also serve as a replacement for the current Pi4J Google Groups moving forward.    Let's try it out and see how it works for us.   You can register here and optionally use your GitHub account to login.  [Pi4J Community Forum](https://forum.pi4j.com/)

thegreystone, 2019-03-07:
Excellent!

thegreystone, 2019-03-08:
I tried creating a new topic under 2.0, but I don't think I have the rights required.

Wanted to check if you'd be interested in doing a talk together at CodeOne 2019 around running Java on the RaspberryPi. :)

savageautomate, 2019-03-08:
@thegreystone 

I put your account in the "Contributors" group after you registered.  That group should have access to create new topics under 2.0.  I wonder if you have to logout and re-login to get the updated permissions applied?    

Sure, I would be interested in a talk.  I have given talks about Pi4J at JavaOne and Devoxx in the past.  Let chat more on the forum.   There is also a "Chat" feature in the forum similar to private messaging.

thegreystone, 2019-03-08:
Hm. Still can't post. Tried from both Safari and Chrome. Chatting seems to work though (at least I seem to be able to send messages).

TangPere, 2019-03-13:
@savageautomate 
I had check the 2.0 branch, for REMOVE JDK INTERNAL REFS I think we can use a JDK tool call jdeps, I will show which code is internal ref, and what can replace it.
Thanks.

thegreystone, 2019-03-13:
We're running on JDK 11 in the Pi4J fork at Robo4J. But there is still much to do.

thegreystone, 2019-03-13:
I'll get the first few changes over into the repo this weekend.

savageautomate, 2019-03-13:
I've got some radical (API breaking) changes planned for the API in v2.  Do you think its best to start with the existing codebase (what's currently in the `develop/2.0` branch) or basically start with a clean slate.

eitch, 2019-03-13:
Maybe starting with a clean slate can be a really good idea, that is if you want to go all-in with modules, which will make a lot of changes in the code anyhow.

FDelporte, 2019-03-18:
I'm working on a blog "series" of Java 11 on PI (see https://webtechie.be/index.php). My next planned step was having a working compiled Java application toggling a led (https://github.com/FDelporte/PiJavaGPIO), with the final goal to have some kind of dashboard application using TilesFX (https://github.com/FDelporte/PiJava). As I was failing with the LED step, I ended up here in this issue.

Anything I can do to help with the JDK 11 "jump"?

savageautomate, 2019-03-19:
@FDelporte 

The following source branch is compiling using JDK11, however, I don't think much (or any) testing has been done.  https://github.com/Pi4J/pi4j/tree/develop/2.0

Snapshot builds can't also be downloaded from the SNAPSHOT repository or from here:
https://pi4j.com/2.0/download.html

Please be aware that major changes (**improvements**) including API changes are slated for Version 2, so you may have to update the blog article from time to time as things change. 

Any help is welcome -- especially in the hardware testing department.  

Thanks, Robert

thegreystone, 2019-03-19:
I am all in favour of clean slate. I don't mind the 2.0 branch being totally broken for a while whilst getting everything in place.

savageautomate, 2019-03-19:
I've been thinking more on this and wanted to get your opinions.  Maybe v1.3 (current master branch) should be basic (read _"minimal effort"_) support for JDK11 and version 2.0 should proceed with a clean slate as proposed and seemingly accepted.

This might allow for immediate/near term JDK11 support while Pi4J v2.0 is likely to take more time getting it all built out properly.  V2.0 is likely to be highly unstable for some time. 

Thanks, Robert

FDelporte, 2019-03-19:
Looking at https://github.com/Pi4J/pi4j/milestone/6 indeed v2 could take some time. So your idea to have a "fast & simple" 11 compliant 1.3 version seems a good approach. What do you think would basically be needed for the 1.3 version?

thegreystone, 2019-03-20:
Are you thinking mrjar?

savageautomate, 2019-03-24:
@thegreystone 

No I was just thinking that maybe a v1.3 release should focus solely on Java 11 compatibility and version 2 could focus on new features and breaking API changes.  So in effect, version 1.3 would mostly be the same feature parity as version 1.2 except for compiled for JDK11.    Thoughts? 

Robert

bootsy52, 2019-06-25:
> I also want to see about changing the open source license for the project in V2. We had to comply with the GPL license in previous versions and force the project to inherit the GPL license because [...]

While I understand the reason behind this, I think this is a huge step backwards for the other platforms which finally got support in pi4j and now being removed. So what about the idea to have other platform added in the form of plugins, in this way the license of the main project can migrated to another license while other platforms remain supported. By the way what is necessary to do to make a platform link dynamically instead of statically? I build the tinkerboard libraries and set it in the build script, however it seems that beside of that pi4j still tries to access libpi4j.so statically.

savageautomate, 2019-07-03:
@bootsy52 

Pi4J v2 will support the concept of multiple platforms but will only implement and test for the Raspberry Pi platform.  Additional platforms can be added via "extensions" or "plugins".  The details have not all been worked out yet, but that is the current plan.

Actually it should be easier to add support for additional platforms going forward as we are also decoupling raw I/O implementation from platform specific bindings.  This means in theory you could potentially add a new platform simply by means of creating a new platform definition file. (assuming the raw I/O implementation is already available)   
 
Thanks, Robert

savageautomate, 2019-07-03:
## IMPORTANT UPDATE 

Here is the versioning strategy moving forward.

* **Version 1.2** - CURRENT RELEASE USING JDK8
(https://github.com/Pi4J/pi4j/tree/release/1.2)     
* **Version 1.3-SNAPSHOT** - LATEST SNAPSHOT BUILD USING JDK8
This branch is intended to include any minor or major changes under the JDK8 builds.
(https://github.com/Pi4J/pi4j/tree/master)     
* **Version 1.4-SNAPSHOT** - LATEST SNAPSHOT BUILD USING JDK11
This branch is **EXPERIMENTAL** and builds using JDK11.
This build has not been fully tested.
(https://github.com/Pi4J/pi4j/tree/develop/1.4)     
* **Version 2.0-SNAPSHOT** - NEXT GENERATION OF Pi4J BUILD ON JDK11
The source code is not yet available for this project.

**Note:** There was a `develop/2.0` branch that has now been removed.  The contents of that branch are now republished as `develop/1.4`.

mrj, 2019-09-02:
It would be worthwhile to add a prominent warning on the download page that 1.2 only fully works on Java 8. The latest Raspbian (Buster) installs with Java 11, and it took me a while to work out why my I2C writes were either hanging or erroring-out.

d-f-clark, 2020-03-21:
Ditto.  And it looks like the sun/misc/SharedSecrets issue has been languishing for a year now.  Is support for this in the 1.3 or 1.4 snapshots?

eitch, 2020-03-22:
I will gladly release a 1.4 if Robert is too busy. I have been using my own 1.4 snapshot fork for a while and would be glad if i could use a release version which was on maven central.

d-f-clark, 2020-03-23:
The README's for 1.3-SNAPSHOT and 1.4-SNAPSHOT make no mention of the sun.misc.SharedSecrets issue with Open JDK 11, or reflectivity access in general.  Rather than wasting time trying them I thought I'd ask the question here.  Do either of these projects fix the sun.misc.SharedSecrets issue, and if so, are there plans to get either of these versions to release?  Thank you

eitch, 2020-03-23:
@d-f-clark i want to get 1.4 released, and will make sure it works on JDK 11+ and certainly 64-bit. I have been thinking of releasing it under my fork with a new groupId, since i don't have write access to this repository and @savageautomate seems to be currently indisposed.

savageautomate, 2020-06-11:
Closing this open issue as the effort is underway.  We (including @eitch ) are actively working on both `v1.4` (https://github.com/Pi4J/pi4j/tree/develop/1.4) and `v2.0` (https://github.com/Pi4J/pi4j-v2) which include support for the newer JDKs.

Thanks, Robert

d-f-clark, 2020-06-12:
Is it typical to close an open issue simply because it’s being worked on?  A typical software development lifecycle includes development of the solution, and testing, before an issue is closed.

savageautomate, 2020-06-12:
@d-f-clark 

You are right ... I got caught up in an effort closing tickets in this V1 repository that are getting addressed in V2 which is a separate repository (https://github.com/Pi4J/pi4j-v2).  However, in this case, the ticket should remain open as there will be a V1.x release to allow compiling using JDK11.  

Let me be clear ... v1.4 can be compiled using JDK11 but only v2.0 will truly supported as a true Java module.

markko1, 2020-09-29:
Hello! I have tried to run 1.4 build on a Raspberry Zero, but I end up with an error from the native library. Is it because of 1.4 is meant for 64 bit and RPI Zero is 32 bit?
Sorry if it's not the right place to ask.

eitch, 2020-09-30:
Hi, i am sorry, but i don't have a pi zero, so i can't test this and i don't know what architecture the pi zero is. If is 32 bit, then it should work, but you still need JDK11, and wiringpi must be installed in the 32bit version.

markko1, 2020-10-01:
Thanks for the answer @eitch ! :)
Yep, Pi Zero is indeed 32 bit, I double checked. I am running [Jdk 11 from Zulu](https://www.azul.com/downloads/zulu-community/?version=java-11-lts&architecture=arm-32-bit-hf&package=jdk) which is a special build for ARMv6. It turned out that Zero is ARMv6 while most of the newer ones are ARMv8 ([link](https://en.wikipedia.org/wiki/Raspberry_Pi#Specifications)).
I'm not sure if this architecture version plays any other role than picking the right JDK build, though. Because Java8 worked well.

I also have wiringpi installed. gpio -v output:
```
gpio version: 2.50
Copyright (c) 2012-2018 Gordon Henderson
This is free software with ABSOLUTELY NO WARRANTY.
For details type: gpio -warranty

Raspberry Pi Details:
  Type: Pi Zero, Revision: 03, Memory: 512MB, Maker: Sony
  * Device tree is enabled.
  *--> Raspberry Pi Zero Rev 1.3
  * This Raspberry Pi supports user-level GPIO access.
```
I started to wonder whether there can be a conflict between my version of wiringpi and the one compiled into PI4J? Not sure if a stupid idea, because I have no clue about how this native code linking works.

I could probably check this idea if I could build PI4J on my Raspberry? Unfortunately I do not know how to do that.
I have built PI4J from develop/1.4 branch using Docker image with instructions from here, but that did not work. This is what I got when running my application with [1.4: link to Github gist](https://gist.github.com/markko1/d09ba3a3e0fbfc37566baf3efe219c19). com.jubla are my packages there :)

markko1, 2020-10-03:
I was able to build Pi4J on my Pi Zero (branch develop/1.4) but unfortunately it made no difference.

FDelporte, 2020-10-03:
For your info @markko1 , I never used a Zero before but your question triggered me to order one :-) will try to reproduce this problem when received

markko1, 2020-11-24:
Hi @FDelporte :)
Did you manage to reproduce the problem?

closed 2021-02-15 by savageautomate

markko1, 2021-02-21:
I can confirm that current master (i.e. 1.4-SNAPSHOT) is working with Raspberry Zero as well!
Very happy now, thanks! :)

FDelporte, 2021-02-22:
Thanks @markko1 for the feedback! Didn't manage to try out myself yet, so nice to have confirmed this issue is solved.