Pi4J/pi4j-v1 #467: WiringPi Deprecated

mrdanimal, 2019-08-15:
Looks like the maintainer of WiringPi got fed up with dealing with jerks and is going to stop maintaining it:
http://wiringpi.com/wiringpi-deprecated/

Is there any plan yet for dealing with this dependency?

savageautomate, 2019-08-15:
Yes, working right now on Pi4J Version 2.0.
Please see this topic:  https://forum.pi4j.com/topic/18/wiringpi-deprecated

We were planning on decoupling from WiringPi anyway in version 2, but not necessarily abandoning it from the lower level access.   With the recent deprecation notice, we are now looking to move over to PIGPIO.  

Sadly, Pi4J was one of the "_culprits_" early on that caused Gordon some frustration.  At first we statically linked to WiringPi to make it "_easier_" for Java programmers coming on board with the Raspberry Pi and not fully understanding shared libs, and external dependencies that were not JARs.  The negative impact of statically linking was that Pi4J could not keep up with the WiringPi releases, especially when new Pi hardware models came out as we did not have early access to new hardware to test and validate against.  In some cases, a low-level error would get thrown by WiringPi which included a note to contact WiringP -- which was wrong as it was really a Pi4J issue and not a WiringPi issue.

In recent releases we moved away from the static linking over to dynamic linking to "fix" this problem but really it was far too late.  We never hounded Gordon for fixes or made any demands on him, but the lag in Pi4J releases did understandably cause him some frustration.  

Our sincerest apologies to Gordon and best wishes to him on his future projects/endeavors.

mrdanimal, 2019-08-15:
Thanks so much for the quick and thorough response. Good to know this
project will continue onward!

Gerry33, 2019-08-18:
Hi Robert and all,
thanks for your efforts on PI4J.

I agree with  @mrdanimal  above, appreciate your efforts to maintain PI4J 
and value your decision to move to PIGPIO.

Perhaps I can add some value here. I'm JAVA-nian now some with some (longer ago) experience in 'C'.

So I may add some help on porting to other platforms than RPI only. 
I'm focused on the ODROIDs, also tried to compile PIGPIO on a ODROID C2, but completely failed. 

So I see a massive problem here when compiling  PIGPIO to other platform.
'WiringPI' has done this approach already, for PIGPIO I never saw any attempt to do so.
There I see the main problem within your decision.

Anyway,  thanks for now
Gerry

savageautomate, 2019-08-18:
@Gerry33 

Check out the new V2 architecture here:
https://forum.pi4j.com/topic/16/version-2-0-architecture

And discussion topics on V2:
https://forum.pi4j.com/category/6/version-2-0

It's still very much a work in progress but it is starting to come together (high level).
I would certainly appreciate any input and of course any efforts to help support other platforms as we move forward.  

My initial focus is the core infrastructure and support for all the RaspberryPi models.  The new plugin architecture will allow for multiple I/O platforms and providers.  So integration with third-party platforms will no longer be tied to any one specific library like WiringPi.   My intention also was to create a generic Linux filesystem integration I/O providers for GPIO/I2C/serial/etc.   This would work for many alternate platforms at the expense of not being the most efficient, but perhaps providing the most coverage.   

I would also prefer third-party integration plugins to be managed and supported outside the core Pi4J project so that would make it easier to track issues and maintain the third-party support without having to make new releases of the core.  

Thanks, Robert

closed 2020-06-11 by savageautomate