Pi4J/pi4j-v1 #468: pi4J Error: illegal reflective access operation OpenJDK11

daabutts, 2019-08-30:
I have narrowed the problem to OpenJDK11.  The code works fine with OpenJDK8 and OracleJDK8.  I got identical results on both pi3 and pi4.  Line causing error:

final ADS1115GpioProvider gpioProvider = new ADS1115GpioProvider(I2CBus.BUS_1, ADS1115GpioProvider.ADS1115_ADDRESS_0x48);

The above Java pi4j command is causing:
<--Pi4J--> ADS1115 GPIO Example ... started.
WARNING: An illegal reflective access operation has occurred
WARNING: Illegal reflective access by com.pi4j.io.file.LinuxFile (file:/home/pi/NBPs/ADS1115/dist/lib/pi4j-core.jar) to field java.nio.Buffer.address
WARNING: Please consider reporting this to the maintainers of com.pi4j.io.file.LinuxFile
WARNING: Use --illegal-access=warn to enable warnings of further illegal reflective access operations
WARNING: All illegal access operations will be denied in a future release
Exception in thread "Thread-1" 
java.lang.NoClassDefFoundError: sun/misc/SharedSecrets
	at com.pi4j.io.file.LinuxFile.getFileDescriptor(LinuxFile.java:215)
	at com.pi4j.io.file.LinuxFile.ioctl(LinuxFile.java:103)
	at com.pi4j.io.i2c.impl.I2CBusImpl.selectBusSlave(I2CBusImpl.java:288)
	at com.pi4j.io.i2c.impl.I2CBusImpl.runBusLockedDeviceAction(I2CBusImpl.java:255)
	at com.pi4j.io.i2c.impl.I2CBusImpl.readBytesDirect(I2CBusImpl.java:146)
	at com.pi4j.io.i2c.impl.I2CDeviceImpl.read(I2CDeviceImpl.java:193)
	at com.pi4j.gpio.extension.ads.ADS1x15GpioProvider$ADCMonitor.run(ADS1x15GpioProvider.java:479)
Caused by: java.lang.ClassNotFoundException: sun.misc.SharedSecrets
	at java.base/jdk.internal.loader.BuiltinClassLoader.loadClass(BuiltinClassLoader.java:583)
	at java.base/jdk.internal.loader.ClassLoaders$AppClassLoader.loadClass(ClassLoaders.java:178)
	at java.base/java.lang.ClassLoader.loadClass(ClassLoader.java:521)
	... 7 more

This error is as a result of running the <--Pi4J--> ADS1115 GPIO Example, see attached.
Also attached log files with system information for Raspbain Stretch and Buster.  Note that the example works with JDK8 but not JDK11.  The same error is given and is included in each file.

[verBusterJDK11.log](https://github.com/Pi4J/pi4j/files/3560864/verBusterJDK11.log)
[verStretchJDK8.log](https://github.com/Pi4J/pi4j/files/3560865/verStretchJDK8.log)
[verBusterJDK8.log](https://github.com/Pi4J/pi4j/files/3560866/verBusterJDK8.log)
[ADS1115.java.txt](https://github.com/Pi4J/pi4j/files/3560869/ADS1115.java.txt)

I assume this will require a pi4j update...

TobiasReich, 2019-10-09:
Yea, I have a similar issue. Is there any fix for that? Looks like not many updates were made these days... :(

eitch, 2019-10-10:
I have a branch working on JDK 11 and arm64:

    https://github.com/eitch/pi4j/commits/develop/1.4

Start your JVM with the following options:

    JAVA_OPTS="--add-exports java.base/jdk.internal.misc=ALL-UNNAMED"

savageautomate, 2019-10-10:
@TobiasReich 

Try testing using the 1.4-SNAPSHOT build.  It is compiled for JDK 11.  
https://github.com/Pi4J/pi4j/tree/develop/1.4

I won't promise the underlying issues were actually fixed, but this branch does at least compile on JDK11.

Thanks, Robert

TobiasReich, 2019-10-12:
Thanks for the hint. I guess I'm doing something wrong.
When I wanted to use the project classes in mine, I noticed that something didn't compile.

com.pi4j.io.file.LinuxFile.java:32 states 

```
// TODO :: REMOVE JDK INTERNAL REFS
import jdk.internal.misc.SharedSecrets;
```

where the SharedSecrets can't be found. Did I miss something / some imports etc.?

eitch, 2019-12-16:
@TobiasReich 

Sorry for the late reply. I should have mentioned that i did the fixes on my personal clone of the project and am waiting for my PR to be merged. In the meantime you can build my project locally:

    git clone https://github.com/eitch/pi4j eitch-pi4j
    cd eitch-pi4j
    # Cross compile:
    RPI_CROSS_COMPILE=true mvn clean install -Prelease -DskipTests -Pall-platforms

And then also use 1.4-SNAPSHOT in your project.

I just tested this on my raspberry pi with Bellsoft JDK 11 installed on my RPi.

kkuez, 2020-08-09:
This is still not fixed on the 1.4 branch.
As well @eitch 
I cant build your repo
` -ePI_CROSS_COMPILE=true mvn clean install -Prelease -DskipTests -Pall-platforms
[INFO] Error stacktraces are turned on.
[INFO] Scanning for projects...
[WARNING] The project com.pi4j:pi4j-parent:pom:1.3-SNAPSHOT uses prerequisites which is only intended for maven-plugin projects but not for non maven-plugin projects. For such purposes you should use the maven-enforcer-plugin. See https://maven.apache.org/enforcer/enforcer-rules/requireMavenVersion.html
[INFO] ------------------------------------------------------------------------
[INFO] Reactor Build Order:
[INFO]
[INFO] Pi4J :: Parent POM                                                 [pom]
[INFO] Pi4J :: JNI Native Library                                         [pom]
[INFO] Pi4J :: Java Library (Core)                                        [jar]
[INFO] Pi4J :: GPIO Extension                                             [jar]
[INFO] Pi4J :: Device Abstractions                                        [jar]
[INFO] Pi4J :: Java Examples                                              [jar]
[INFO] Pi4J :: Distribution                                               [deb]
[INFO]
[INFO] ------------------------< com.pi4j:pi4j-parent >------------------------
[INFO] Building Pi4J :: Parent POM 1.3-SNAPSHOT                           [1/7]
[INFO] --------------------------------[ pom ]---------------------------------
[INFO]
[INFO] --- maven-clean-plugin:2.5:clean (default-clean) @ pi4j-parent ---
[INFO] Deleting D:\Muell\eitch-pi4j\target
[INFO]
[INFO] --- license-maven-plugin:1.14:update-file-header (append-license-file-headers) @ pi4j-parent ---
[WARNING] No file to scan.
[INFO]
[INFO]
[INFO] --- license-maven-plugin:1.14:update-project-license (update-project-license) @ pi4j-parent ---
[INFO]
[INFO] --- whitespace-maven-plugin:1.0.4:trim (default) @ pi4j-parent ---
[INFO]
[INFO] --- build-helper-maven-plugin:3.0.0:attach-artifact (attach-artifacts) @ pi4j-parent ---
[INFO]
[INFO] --- maven-javadoc-plugin:2.10.4:jar (attach-javadocs) @ pi4j-parent ---
[INFO] Not executing Javadoc as the project is not a Java classpath-capable package
[INFO]
[INFO] --- maven-gpg-plugin:1.6:sign (sign-artifacts) @ pi4j-parent ---
gpg: no default secret key: No secret key
gpg: signing failed: No secret key
[INFO] ------------------------------------------------------------------------
[INFO] Reactor Summary for Pi4J :: Parent POM 1.3-SNAPSHOT:
[INFO]
[INFO] Pi4J :: Parent POM ................................. FAILURE [  1.410 s]
[INFO] Pi4J :: JNI Native Library ......................... SKIPPED
[INFO] Pi4J :: Java Library (Core) ........................ SKIPPED
[INFO] Pi4J :: GPIO Extension ............................. SKIPPED
[INFO] Pi4J :: Device Abstractions ........................ SKIPPED
[INFO] Pi4J :: Java Examples .............................. SKIPPED
[INFO] Pi4J :: Distribution ............................... SKIPPED
[INFO] ------------------------------------------------------------------------
[INFO] BUILD FAILURE
[INFO] ------------------------------------------------------------------------
[INFO] Total time:  2.554 s
[INFO] Finished at: 2020-08-09T11:58:31+02:00
[INFO] ------------------------------------------------------------------------
[ERROR] Failed to execute goal org.apache.maven.plugins:maven-gpg-plugin:1.6:sign (sign-artifacts) on project pi4j-parent: Exit code: 2 -> [Help 1]
org.apache.maven.lifecycle.LifecycleExecutionException: Failed to execute goal org.apache.maven.plugins:maven-gpg-plugin:1.6:sign (sign-artifacts) on project pi4j-parent: Exit code: 2
    at org.apache.maven.lifecycle.internal.MojoExecutor.execute (MojoExecutor.java:215)
    at org.apache.maven.lifecycle.internal.MojoExecutor.execute (MojoExecutor.java:156)
    at org.apache.maven.lifecycle.internal.MojoExecutor.execute (MojoExecutor.java:148)
    at org.apache.maven.lifecycle.internal.LifecycleModuleBuilder.buildProject (LifecycleModuleBuilder.java:117)
    at org.apache.maven.lifecycle.internal.LifecycleModuleBuilder.buildProject (LifecycleModuleBuilder.java:81)
    at org.apache.maven.lifecycle.internal.builder.singlethreaded.SingleThreadedBuilder.build (SingleThreadedBuilder.java:56)
    at org.apache.maven.lifecycle.internal.LifecycleStarter.execute (LifecycleStarter.java:128)
    at org.apache.maven.DefaultMaven.doExecute (DefaultMaven.java:305)
    at org.apache.maven.DefaultMaven.doExecute (DefaultMaven.java:192)
    at org.apache.maven.DefaultMaven.execute (DefaultMaven.java:105)
    at org.apache.maven.cli.MavenCli.execute (MavenCli.java:957)
[... 17 lines cut ...]
    at org.apache.maven.lifecycle.internal.MojoExecutor.execute (MojoExecutor.java:148)
    at org.apache.maven.lifecycle.internal.LifecycleModuleBuilder.buildProject (LifecycleModuleBuilder.java:117)
    at org.apache.maven.lifecycle.internal.LifecycleModuleBuilder.buildProject (LifecycleModuleBuilder.java:81)
    at org.apache.maven.lifecycle.internal.builder.singlethreaded.SingleThreadedBuilder.build (SingleThreadedBuilder.java:56)
    at org.apache.maven.lifecycle.internal.LifecycleStarter.execute (LifecycleStarter.java:128)
    at org.apache.maven.DefaultMaven.doExecute (DefaultMaven.java:305)
    at org.apache.maven.DefaultMaven.doExecute (DefaultMaven.java:192)
    at org.apache.maven.DefaultMaven.execute (DefaultMaven.java:105)
    at org.apache.maven.cli.MavenCli.execute (MavenCli.java:957)
[... cut ...]
    at org.codehaus.plexus.classworlds.launcher.Launcher.launch (Launcher.java:225)
    at org.codehaus.plexus.classworlds.launcher.Launcher.mainWithExitCode (Launcher.java:406)
    at org.codehaus.plexus.classworlds.launcher.Launcher.main (Launcher.java:347)
[ERROR]
[ERROR] Re-run Maven using the -X switch to enable full debug logging.
[ERROR]
[ERROR] For more information about the errors and possible solutions, please read the following articles:
[ERROR] [Help 1] http://cwiki.apache.org/confluence/display/MAVEN/MojoExecutionException
`

FDelporte, 2020-08-10:
PR https://github.com/Pi4J/pi4j/pull/498 has been merged
FYI building the 1.4 branch is easiest with Docker, by using 
```
docker pull pi4j/pi4j-builder:1.4
docker run --user "$(id -u):$(id -g)" --rm --volume $(pwd):/build pi4j/pi4j-builder:1.4
```

eitch, 2020-08-11:
I would suggest building from the root of the project using the following:

    mvn clean install -DskipTests -P raspberrypi

This should work on the develop/1.4 branch, as i am constantly doing it, trying to fix a native bug.

eitch, 2020-08-31:
I think we can close this issue, as develop/1.4 works well on JDK11+

closed 2020-08-31 by eitch

tatery, 2020-09-21:
@eitch how did you manage to have it working? I'm using ver 1.4-SNAPSHOT (downloaded from here: https://oss.sonatype.org/content/groups/public/com/pi4j/pi4j-core/) and all the time have this error:
`java.lang.IllegalAccessError: class com.pi4j.io.file.LinuxFile (in unnamed module @0x127a7e) cannot access class jdk.internal.misc.SharedSecrets (in module java.base) because module java.base does not export jdk.internal.misc to unnamed module @0x127a7e
        at com.pi4j.io.file.LinuxFile.getFileDescriptor(LinuxFile.java:183) ~[pi4j-core-1.4-SNAPSHOT.jar:?]
        at com.pi4j.io.file.LinuxFile.ioctl(LinuxFile.java:71) ~[pi4j-core-1.4-SNAPSHOT.jar:?]
        at com.pi4j.io.i2c.impl.I2CBusImpl.selectBusSlave(I2CBusImpl.java:289) ~[pi4j-core-1.4-SNAPSHOT.jar:?]
        at com.pi4j.io.i2c.impl.I2CBusImpl.runBusLockedDeviceAction(I2CBusImpl.java:256) ~[pi4j-core-1.4-SNAPSHOT.jar:?]
        at com.pi4j.io.i2c.impl.I2CBusImpl.readByte(I2CBusImpl.java:151) ~[pi4j-core-1.4-SNAPSHOT.jar:?]
        at com.pi4j.io.i2c.impl.I2CDeviceImpl.read(I2CDeviceImpl.java:203) ~[pi4j-core-1.4-SNAPSHOT.jar:?] 
`

tatery, 2020-09-22:
@eitch finally it is working for me as well. I had to add the following configuration to my project pom.xml file:

```xml
<plugin>
    <groupId>org.apache.maven.plugins</groupId>
    <artifactId>maven-surefire-plugin</artifactId>
    <configuration>
        <argLine>--add-exports java.base/jdk.internal.misc=ALL-UNNAMED</argLine>
    </configuration>
</plugin>
```

eitch, 2020-09-22:
Hey @tatery glad to hear that!

Cawdeen, 2020-11-24:
I found myself here as I have the exact same error. I am currently using Pi4j 1.2. Can someone be kind enough to let me know I go about downloading either 1.4 or 2.0 in the snapshot above? Do I download all the files individually and then create the library in NetBeans adding each file on at a time?