Pi4J/pi4j-v1 #469: package com.pi4j does not exist

amastrobera, 2019-09-03:
Dear Pi4J, 

thanks for your library. I had tried the [python gpio](https://pypi.org/project/RPi.GPIO/) before, but I have to make the same thing in Java now.  I cannot compile the code when using `wiringpi` module. **can you help me build it?**

My code (Test.java)
```
package test;

import com.pi4j.wiringpi;

public class Test {
  public static void main(String[] args) throws InterruptedException {
   
    if (wiringpi.Gpio.wiringPiSetup() == -1) {
      System.out.println(" ==>> GPIO SETUP FAILED");
      throw new Exception("Gpio.wiringPiSetup() failed");
    }
    System.out.println("<--Pi4J--> GPIO Control Example ... started.");
  
    wiringpi.Gpio.piBoardRev();
    wiringpi.Gpio.pinMode(7, wiringpi.Gpio.OUTPUT);
    wiringpi.Gpio.pinMode(11, wiringpi.Gpio.OUTPUT);
    wiringpi.Gpio.pinMode(13, wiringpi.Gpio.OUTPUT);
    wiringpi.Gpio.pinMode(15, wiringpi.Gpio.OUTPUT);

    System.out.println("    > backward");
    wiringpi.Gpio.digitalWrite(7, wiringpi.Gpio.HIGH);
    wiringpi.Gpio.digitalWrite(11, wiringpi.Gpio.LOW);
    wiringpi.Gpio.digitalWrite(13, wiringpi.Gpio.HIGH);
    wiringpi.Gpio.digitalWrite(15, wiringpi.Gpio.LOW);
   
   // bla bla bla some more code 

  }
}
```
Structure of my project
main_dir
   - classes 
   - libs
       - pi4j-core.jar
   - test
       - Test.java

I have downloaded the `jar` files into a lib folder aside the test/Test.java, so I compile it like this (I also compile other stuff depending on jar files this way, and works ... but not wiringpi)

```
javac -cp classes:/libs/pi4j-core.jar -d classes test/Test.java
```

Error : 
```
/path/to/Test.java:xx: error: package com.pi4j does not exist 
import com.pi4j.wiringpi; 
               ^                      
/path/to/Test.java:xx: error: package wiringpi does not exist/opt/pi4j
    if (wiringpi.Gpio.wiringPiSetup() == -1) { 
               ^                      
// etc etc for all calls 
```

amastrobera, 2019-09-03:
@savageautomate  and @mrdanimal
does this have anything to do with [wiringpi being deprecated](https://github.com/Pi4J/pi4j/issues/467) ?

savageautomate, 2021-01-11:
**Closed**.  

`import com.pi4j.wiringpi;` is just the package name.  You need to import the actual class you want to use or import all classes from the package using `import com.pi4j.wiringpi.*;`
...
The following works for me:

```
import com.pi4j.wiringpi.Gpio;

public class Test {
    public static void main(String[] args) throws Exception {

        if (Gpio.wiringPiSetup() == -1) {
            System.out.println(" ==>> GPIO SETUP FAILED");
            throw new Exception("Gpio.wiringPiSetup() failed");
        }
        System.out.println("<--Pi4J--> GPIO Control Example ... started.");

        Gpio.piBoardRev();
        Gpio.pinMode(7, Gpio.OUTPUT);
        Gpio.pinMode(11, Gpio.OUTPUT);
        Gpio.pinMode(13, Gpio.OUTPUT);
        Gpio.pinMode(15, Gpio.OUTPUT);

        System.out.println("    > backward");
        Gpio.digitalWrite(7, Gpio.HIGH);
        Gpio.digitalWrite(11, Gpio.LOW);
        Gpio.digitalWrite(13, Gpio.HIGH);
        Gpio.digitalWrite(15, Gpio.LOW);

        // bla bla bla some more code
    }
}

```

closed 2021-01-11 by savageautomate