Pi4J/pi4j-v1 #470: Can't build dex file from pi4j-core.jar

BotovNikita, 2019-10-18:
I try to deploy pi4j to Android project with FireMonkey in Rad Studio XE 10.2.3.
I use dx.bat to convert jar to dex. All jar's convert ok, but pi4j-core.jar have error:
`C:\Users\Botov\AppData\Local\Android\Sdk\build-tools\29.0.2>dx --dex --output=D:\DEX_MAKER\output\dex\pi4j-core.dex --positions=lines D:\DEX_MAKER\jar\pi4j-core.jar
Uncaught translation error: com.android.dx.cf.code.SimException: ERROR in com.pi4j.io.file.LinuxFile.mmap:(ILcom/pi4j/io/file/LinuxFile$MMAPProt;Lcom/pi4j/io/file/LinuxFile$MMAPFlags;I)Ljava/nio/ByteBuffer;: invalid opcode ba - invokedynamic requires --min-sdk-version >= 26 (currently 13)
Uncaught translation error: com.android.dx.cf.code.SimException: ERROR in com.pi4j.io.i2c.impl.I2CBusImpl.readByteDirect:(Lcom/pi4j/io/i2c/I2CDevice;)I: invalid opcode ba - invokedynamic requires --min-sdk-version >= 26 (currently 13)
2 errors; aborting`

What can I do?

BotovNikita, 2019-10-18:
I make it. Jast add --min-sdk-version=26 at start of command line, and it's works.

But now, when I use the lib, I am have an error:
java.lang.UnsatisfiedLinkError: No implementation found for int con.pi4j.wiringpi.Gpio.wiringPiSetup()
(tried Java_com_pi4j_wiring_Gpio_wiringPiSetup and Java_com_pi4j_wiring_Gpio_wiringPiSetup_)

My code like:
```
....
  gpio : JGpioController;
  pin :  JGpioPinDigitalOutput;
begin
  gpio := TJGpioFactory.JavaClass.getInstance;
...
```

savageautomate, 2020-06-11:
Embedded inside the Pi4J Core JAR file is a native library file (`libpi4j.so`) that gets extracted at runtime and loaded into the JVM.  I suspect in the case of Android something is going wrong with extracting and loading this native library.  You may have to include the library file manually and include it as part of the java libraries path.

closed 2020-06-11 by savageautomate