Pi4J/pi4j-v1 #471: GPIO Digital Inputs with Internal Pull-up on Raspberry PI 4

dpotts-sc, 2019-10-25:
I have code that has been working great on the Raspberry PI3; I just assigned a GPIO as a digital input (Raspberry PI GPIO21, pin 40, WiringPi 29) with a pull-up on it, and then my switch pulls it down to ground when it is pressed.

I've upgraded to a PI4 and now this code doesn't work.  If I move to GPIO2 (WiringPi 8, Pin 3), it works fine, but this is not ideal due to the additional wiring and various devices connected to the PI.  

I'm guessing it has to do with WiringPi not fully upgraded to support PI4 and / or Java.  Or maybe Pi4J not updated yet for the 4.  Thoughts?

dpotts-sc, 2019-10-25:
Some additional information, SOME of the pins are usable with pullups on them.  The all seem to work if they are Push-Pull.

Pin | Name | Good? |   | Pin | Name | Good?
-- | -- | -- | -- | -- | -- | --
1 | 3.3v | -- |   | 2 | 5v | --
3 | GPIO2 | Yes |   | 4 | 5v | --
5 | GPIO3 | Yes |   | 6 | gnd | --
7 | GPIO4 | Yes |   | 8 | GPIO14 | No
9 | gnd | -- |   | 10 | GPIO15 | Yes
11 | GPIO17 | No |   | 12 | GPIO18 | No
13 | GPIO27 | No |   | 14 | gnd | --
15 | GPIO22 | No |   | 16 | GPIO23 | No
17 | 3.3v | -- |   | 18 | GPIO24 | No
19 | GPIO10 | No |   | 20 | gnd | --
21 | GPIO9 | No |   | 22 | GPIO25 | No
23 | GPIO11 | No |   | 24 | GPIO8 | Yes
25 | gnd | -- |   | 26 | GPIO7 | Yes
27 | ID_SC | -- |   | 28 | ID_SC | --
29 | GPIO5 | Yes |   | 30 | gnd | --
31 | GPIO6 | Yes |   | 32 | GPIO12 | No
33 | GPIO13 | No |   | 34 | gnd | --
35 | GPIO19 | No |   | 36 | GPIO16 | No
37 | GPIO26 | No |   | 38 | GPIO20 | No
39 | gnd | -- |   | 40 | GPIO21 | No

dpotts-sc, 2019-10-25:
I seem to have found a temporary work-around.  Hopefully this will help someone. 

In the /boot/config.txt file you can define what the default states are for the GPIO.  I went in and changed the default for the GPIO pins I needed and added the PullUp to the pin there.  This appears to be working now.  

Doesn't fix the fundamental problem, but it can keep me moving forward.  I'm guessing it could be the reason I don't see the changes to the interrupts coming in as well.

BTW, I wrote a walking test that goes pin by pin and displays the status of every GPIO and then turns on/off the pull up / down on each one and on the PI4, it is clear that NONE of the PU/PDs work at all.  The same test works perfectly on the PI3.

matthiesenj, 2019-11-25:
Configuring pull-up/-down on Pi4 works differently than on the Pi3 (other registers, etc), so you'll need wiringPi updated to support that, but as far as I know, it's no longer maintained!?

pablo67340, 2019-12-12:
Any solutions on this yet? I am having similar problems with reading a DHT11 sensor. Unable to do so

dpotts-sc, 2019-12-12:
Not directly, this is a problem with Wiring PI and it doesn't appear that there will be any updates with it anymore.  So we are, well screwed.   

If you don't require changing the PINs from input to output you can use the method I ended up using, changing the /boot/config.txt file so that some pins are inputs, some are outputs.  Then I was able to successfully use the pins but I cannot programmatically change their functionality.

pablo67340, 2019-12-13:
Unfortunately I do need to set the pin between input/output so your workaround will not work for me. For some reason it works fine in Python, just not Pi4J

Maddinthebrain, 2020-06-01:
I just ran into the same problem with the pullup resistors, I guess there is now solution yet, right??

valdar, 2020-06-01:
@Maddinthebrain compile a new version of wiringpi from here https://github.com/WiringPi/WiringPi and use that one should be a good start.

Maddinthebrain, 2020-06-01:
well I have the original version 2.52. It doesn't with it.

eitch, 2020-08-31:
@dpotts-sc @Maddinthebrain @pablo67340 can you guys try my fork of wiringPI: https://github.com/eitch/wiringPi or at least try the latest https://github.com/WiringPi/WiringPi version 2.60 and see if the issue persists? I have been using an I2C bus with interrupts on GPIO BCM pins 9, 10, 22, 23, 24 and have not had any issues. I am on aarch64 with my wiringPi branch.

savageautomate, 2021-01-14:
Issue confirmed.
WiringPi v2.52 added support for the RPi 4B but does not include changes to properly handle the pin PULL UP/DOWNs.   

Using the _unofficial_ WiringPi v2.60 (https://github.com/WiringPi/WiringPi ) as @eitch suggested does resolve this problem.
I have added comments in the site documentation for the Pi4J v1.3 release as well as an installer script to make this as easy as possible.   

See:
 - https://pi4j.com/1.3/install.html#WiringPi_Native_Library
 - https://webtechie.be/post/2020-10-10-pi4j-on-raspberry-pi-4/

closed 2021-01-14 by savageautomate