Pi4J/pi4j-v1 #473: final GpioController gpio = GpioFactory.getInstance(); hangs on RPi4

izhidov, 2019-11-21:
Is RPi4 supported in Pi4j? I dont see it mentioned anywhere in the docs and I have am issue where 
 final GpioController gpio = GpioFactory.getInstance();  just hangs and never comes back.

izhidov, 2019-11-21:
Found the culprit but whats the resolution?:
Unable to determine hardware version. I see: Hardware   : BCM2835
,
 - expecting BCM2708 or BCM2709.
If this is a genuine Raspberry Pi then please report this
to projects@drogon.net. If this is not a Raspberry Pi then you
are on your own as wiringPi is designed to support the
Raspberry Pi ONLY.

izhidov, 2019-11-21:
WiringPi has been updated to 2.52, still same issue

gpio version: 2.52
Copyright (c) 2012-2018 Gordon Henderson
This is free software with ABSOLUTELY NO WARRANTY.
For details type: gpio -warranty

Raspberry Pi Details:
  Type: Pi 4B, Revision: 01, Memory: 4096MB, Maker: Sony 
  * Device tree is enabled.
  *--> Raspberry Pi 4 Model B Rev 1.1
  * This Raspberry Pi supports user-level GPIO access.

eitch, 2020-08-31:
Hi @izhidov i've been running on a RPi4 with wiringPi 2.60 for quite a while without issues. Does gpio readall work? Are you running everything with a normal user or root? Are you using Java 11+?

izhidov, 2020-08-31:
I decided to wait till 2.0 comes out.

closed 2020-08-31 by eitch