Pi4J/pi4j-v1 #475: Rpi4 support

mayurirane-sipl, 2019-12-17:
I am using this lib for rpi3B+ now want to use it for rpi4 for handling gpios. But its is not working neither any error is thrown.

ShisanXing, 2019-12-19:
This project may have stoped I guess, will pi4j-v2 support jdk8?

rolaca11, 2019-12-20:
I got this to work on my Rpi4b, but the GPIO addresses are way off. It would be nice to support the new revisions of the pi

rolaca11, 2019-12-20:
sorry, my fault. I had to update my WiringPI to 2.5.2 and then with `gpio readall` command, I could get a better picture about the addresses of the GPIO ports

rolaca11, 2019-12-20:
I'd add that me finding about "wrong" GPIO addresses was down to pure luck. You could take a look, if you addressed you GPIO ports based on the following diagram:
[image]

where the name/wPi columns is what you are looking for

FDelporte, 2019-12-22:
Just a tip, I'm working on a Maven library which contains Raspberry Pi version history and pinning info. It can help you to identify pin numbers inside Java similar to "gpio readall". More info on:

https://webtechie.be/2019/11/13/raspberry-pi-history-versions-pins-headers-as-a-java-maven-library

jjzabkar, 2020-01-07:
> I got this to work on my Rpi4b, but the GPIO addresses are way off. It would be nice to support the new revisions of the pi

@rolaca11 Would this library help? https://github.com/FDelporte/PiHeaders

mbates14, 2020-02-06:
Not sure what you did, but it doesnt work at all for me.

mbates14, 2020-02-06:
Actually, it works perfect as long as my Jar isnt running as sudo. But if I switch to root, or run as sudo, no go. 

What??

Laszlo34, 2020-02-18:
This thread is very confusing to me.
So does v1.2 support the RPi 4?
I just got the rpi 4, my first ever. I'm very excited... except I'm finding Pi4j isn't working for me. The release notes don't mention support for it.  But @mbates14 and @FDelporte, you guys sound almost like it is working for you.
<scratching head>

FDelporte, 2020-02-18:
I have been using Pi4J with a Pi3 for my book https://leanpub.com/gettingstartedwithjavaontheraspberrypi/ and also found out I²C didn't work as expected with Java 11.... So unfortunately can not give you more info for the Pi4. Did you update WiringPi to the latest version? That can already solve some of your problems I've read. I also used Arduino and Python scripts combined with Java applications where I found missing functionality within Java...

Laszlo34, 2020-02-19:
> Did you update WiringPi
I'm too new to Pi to know what that means. :D
I'll look into it.

Laszlo34, 2020-02-19:
I'm still having problems. Things are just kinda 'off'...

Here is my gpio -v:

>  gpio -v
gpio version: 2.52
Copyright (c) 2012-2018 Gordon Henderson
This is free software with ABSOLUTELY NO WARRANTY.
For details type: gpio -warranty

Raspberry Pi Details:
  Type: Pi 4B, Revision: 01, Memory: 4096MB, Maker: Sony 
  * Device tree is enabled.
  *--> Raspberry Pi 4 Model B Rev 1.1
  * This Raspberry Pi supports user-level GPIO access.

So I think my WiringPi is 2.52 and up to date for the Pi4.

What I'm seeing is, I connct my 'hello world' LED to GPIO21 on 40 and to GND on pin 34 (220ohm resistor). Then I use the following method to blink it:

`  public static void singlePinTest() throws Exception
  {
    final GpioController gpio = GpioFactory.getInstance();
    final GpioPinDigitalOutput pin = gpio.provisionDigitalOutputPin(GPIO_21, PinState.HIGH);
    pin.setShutdownOptions(true, PinState.LOW);

    // This should blink the LED twice per second for 10 seconds.
    for(int i=0;i<10;i++)
    {
      System.out.println("setting high");
      pin.high();
      Thread.sleep(250);
      System.out.println("setting low");
      pin.low();
      Thread.sleep(250);
    }
    gpio.shutdown();
  }`

Unfortunately, that doesn't appear to do anything at all.

BUT, I did some experimentation, and it turns out, if I change my code to address GPIO_29, which according to pinout doesn't _exist,_ then it blinks exactly as desired.

That can't be right!! At least I'm pretty sure my hardware setup and code are _basically_ working.

But... I must be missing something!

Laszlo34, 2020-02-19:
$ pinout
(...snip...)
J8:
   3V3  (1) (2)  5V    
 GPIO2  (3) (4)  5V    
 GPIO3  (5) (6)  GND   
 GPIO4  (7) (8)  GPIO14
   GND  (9) (10) GPIO15
GPIO17 (11) (12) GPIO18
GPIO27 (13) (14) GND   
GPIO22 (15) (16) GPIO23
   3V3 (17) (18) GPIO24
GPIO10 (19) (20) GND   
 GPIO9 (21) (22) GPIO25
GPIO11 (23) (24) GPIO8 
   GND (25) (26) GPIO7 
 GPIO0 (27) (28) GPIO1 
 GPIO5 (29) (30) GND   
 GPIO6 (31) (32) GPIO12
GPIO13 (33) (34) GND     <--- my LED ground is here
GPIO19 (35) (36) GPIO16
GPIO26 (37) (38) GPIO20
   GND (39) (40) GPIO21    <--- my LED positive is here

FDelporte, 2020-02-19:
Without looking into your code, but make sure you use wiringpi number and not bcm number when you call the pi4j methods...

[image]

Laszlo34, 2020-02-19:
That's....really crazy! heehe! The sequence isn't even contiguous.

Is there some documentation for this? I see how physical pin 40 says WPI 29. So this is at least hugely helpful. Thank you!! But wow, comparing this to the Java constants in pi4j really has me scratching my head!  :D

FDelporte, 2020-02-19:
don't want to abuse this forum, but read my book for full description ;-)
https://leanpub.com/gettingstartedwithjavaontheraspberrypi/

Laszlo34, 2020-02-19:
You're absolutely right. I'm sorry. This was a issue report, not a help forum! Thanks for the link and info!

savageautomate, 2021-01-14:
I realize this is long overdue, but I have fully tested the GPIO inputs and outputs on RPi4B for the upcoming Pi4J v1.3 release.  Everything is working properly and the Pi4J v1.3-SNAPSHOT build/artifacts are available now and the full release is coming in the next couple of weeks.  Once caveat however, you must install the _unofficial_ v2.60 of the WiringPi library available here: https://github.com/WiringPi/WiringPi

See:
 - https://pi4j.com/1.3/install.html#WiringPi_Native_Library
 - https://webtechie.be/post/2020-10-10-pi4j-on-raspberry-pi-4/

closed 2021-01-14 by savageautomate

savageautomate, 2021-01-14:
> That's....really crazy! heehe! The sequence isn't even contiguous.
> 
> Is there some documentation for this? 

Yes.  
 
 - https://pi4j.com/1.3/pin-numbering-scheme.html
 - http://wiringpi.com/pins/

> I see how physical pin 40 says WPI 29. So this is at least hugely helpful. Thank you!! But wow, comparing this to the Java constants in pi4j really has me scratching my head! :D

You can also use the direct Broadcom pin numbers if you prefer ...
Example:
 - https://github.com/Pi4J/pi4j/blob/master/pi4j-example/src/main/java/BroadcomPinNumberingGpioExample.java
 
```java
// in order to use the Broadcom GPIO pin numbering scheme, we need to configure the
// GPIO factory to use a custom configured Raspberry Pi GPIO provider
GpioFactory.setDefaultProvider(new RaspiGpioProvider(RaspiPinNumberingScheme.BROADCOM_PIN_NUMBERING));

// create gpio controller
final GpioController gpio = GpioFactory.getInstance();

// provision broadcom gpio pin #02 as an output pin and turn on
final GpioPinDigitalOutput pin = gpio.provisionDigitalOutputPin(RaspiBcmPin.GPIO_02, "MyLED", PinState.HIGH);
```