Pi4J/pi4j-v1 #478: Errors Flood Console on JVM Terminate

FidgetyRat, 2020-02-29:
I'm running a project on a Raspberry Pi 3B+ using Pi4j v1.2. While the program is running, everything seems to be working fine, GPIOs trigger, change notifications occur fine, etc.

The problem is, when I quit my application with CTRL+C my console is flooded with thousands of the following errors. These `read(): No such device` errors grow in size as the program runs. The following was from about 5 minutes of execution.

`
**... <Regular console output appears here - I omitted> ...**
2020-02-29 10:41:29 INFO  MessageServer - Connection to client xxx.xxx.xxx.xxx:63986 is no longer active. Cleaning up.
^C **<Here is my CTRL + C quit signal>**
read(): No such device
read(): No such device
**... Thousands of these errors ...**
read(): No such device
read(): No such device
read(): No such device
read(): No such device
read(): No such device
read(): No such device
Exception in thread "Thread-110"
java.util.ConcurrentModificationException
	at java.util.ArrayList$Itr.checkForComodification(ArrayList.java:909)
	at java.util.ArrayList$Itr.next(ArrayList.java:859)
	at com.pi4j.io.gpio.GpioProviderBase.dispatchPinDigitalStateChangeEvent(GpioProviderBase.java:354)
	at com.pi4j.io.gpio.WiringPiGpioProviderBase.pinStateChange(WiringPiGpioProviderBase.java:290)
	at com.pi4j.wiringpi.GpioInterrupt.pinStateChangeCallback(GpioInterrupt.java:128)
read(): No such device
**... Thousands of these errors ...**
read(): No such device
pi@remote:~/$
`

Is there a way to disable these errors in the event they don't mean anything. I'm afraid that they are being cached in memory and flushed when the JVM quits which will eventually fill up the Pi's relatively small memory.

FidgetyRat, 2020-03-01:
Just a bit of research:

I updated to the 1.3 snapshot and there is no change.
I backported wiringpi from 2.50 to 2.46 which was out around the same time and is compatible with the Pi3b+ and there was no change.

I commented out the following shutdown options in my code and the problem went away
`sensor.setShutdownOptions(true, PinState.LOW, PinPullResistance.PULL_DOWN, PinMode.DIGITAL_INPUT);`
So I suspect it has something to do with the shutdown options.

savageautomate, 2021-01-14:
Same issue reported in #457 

Issue has been fixed for upcoming v1.3 release and is available now in v1.3-SNAPSHOT.

closed 2021-01-14 by savageautomate