Pi4J/pi4j-v1 #479: Threads created on each event never terminate?

qzic, 2020-03-06:
I have a bunch of digital listeners on a set of gpio pins. They fire appropriately but each event creates a 2 new threads (high->low and low->high) which never terminate. Should they not terminate after the event handler returns? This happens even if the event handler does nothing but return. Is this a bug or is there some solution?

eitch, 2020-07-27:
@qzic I have just added a change to perform single threaded GPIO event firing on my branch ( https://github.com/Pi4J/pi4j/issues/489 ). While checking the code i am pretty sure that what you are seeing is just the default cached thread pool, which would only stop the thread if it deems it necessary. I don't think this is an issue and would close it. If you prefer less threads, then use the single threaded executor service

eitch, 2020-08-31:
I have also seen that there was a situation where the native C thread binds to the JVM on every interrupt, this has been changed as well, so you might also have seen this happening. Closing now, as the binding is now only done once on registering of the first interrupt handler.

closed 2020-08-31 by eitch

savageautomate, 2021-01-11:
@eitch 's changes are in the v1.4 branch.  Moving this issue to v1.4 for tracking purposes.