Pi4J/pi4j-v1 #483: Pi4J does not build on OpenJDK

stefanofornari, 2020-03-15:
When building on OpenJDK (master and 1.2), the following errors are thrown:

```
[INFO] -------------------------------------------------------------
[ERROR] COMPILATION ERROR : 
[INFO] -------------------------------------------------------------
[ERROR] /home/ste/Projects/pi4j/pi4j-core/src/main/java/com/pi4j/io/file/LinuxFile.java:[34,16] cannot find symbol
  symbol:   class Cleaner
  location: package sun.misc
[ERROR] /home/ste/Projects/pi4j/pi4j-core/src/main/java/com/pi4j/io/file/LinuxFile.java:[35,16] cannot find symbol
  symbol:   class SharedSecrets
  location: package sun.misc
[ERROR] /home/ste/Projects/pi4j/pi4j-core/src/main/java/com/pi4j/io/file/LinuxFile.java:[215,32] cannot find symbol
  symbol:   variable SharedSecrets
  location: class com.pi4j.io.file.LinuxFile
[ERROR] /home/ste/Projects/pi4j/pi4j-core/src/main/java/com/pi4j/io/file/LinuxFile.java:[313,15] cannot find symbol
  symbol:   class Cleaner
  location: class com.pi4j.io.file.LinuxFile
[INFO] 4 errors 
```
I guess it has to do with the fact that OpenJDK misses some internal sun packages.

Any chances this gets fixed?

eitch, 2020-03-16:
Hi @stefanofornari are you building using Java 11+? I would recommend building branch develop/1.4. If you need 64-bit, then try my fork https://github.com/eitch/pi4j/tree/develop/1.4

The issues you are having is not because of the OpenJDK but rather because of Java 11+ restrictions on those classes.

eitch, 2020-03-16:
I should probably also say that my branch is prepared for aarch64 and is not tested on armhf systems.

eitch, 2020-03-16:
If you do test it or armhf and find issues i will gladly fix them.

stefanofornari, 2020-03-17:
Hello, I actually realized I am building with Oracle JDK 10 on Ubuntu. With Java 11, develop/1.4 builds but with tests failing. I guess this is not good ... :( Looking at the code, I see the offending classes have been removed, so it builds with Java 10 too.

I am not sure I understood with

> If you do test it or armhf and find issues i will gladly fix them.

Is there a test suit? you mean run it on a Raspberry? let me know, I'll be happy to help.

eitch, 2020-03-17:
Yes, i do mean running on a RPi. I have not run the tests, but just used the codebase extensively in my projects without issues at the moment. I do use a reduced set of functionality: I2C and GPIOs. With GPIOs i register for events on inputs and write outputs and have not seen any issues yet.

stefanofornari, 2020-03-17:
Ok, I will give it a try and let you know if I notice any issues. In my project I am using pi4j just to read an input pin, so not extensive testing.

stefanofornari, 2020-04-05:
Hi, for the purpose I had it works. It basically listen to incoming event from a PIR sensor.

(see https://github.com/stefanofornari/falco for details)

savageautomate, 2020-06-11:
The `master` branch and `release/1.2` require building using JDK 1.8.  Not just the compiler version flags, but the actual JDK 1.8 installed.   Pi4J v1.4 and newer add support building on newer JDKs.

closed 2020-06-11 by savageautomate