Pi4J/pi4j-v1 #491: Unable to load [libpi4j.so] using path: [/lib/raspberrypi/dynamic/libpi4j.so]

pavankuppa, 2020-06-10:
It seems the path is not loading which is present inside the jar file

Installed wiringpi too - ```sudo apt-get install wiringpi```
```sh
$ ldconfig -p | grep wiringPi.so
libwiringPi.so.2 (libc6,hard-float) => /lib/arm-linux-gnueabihf/libwiringPi.so.2
```

Even I extract the jar and keep libpi4j.so in lib folder as(raspberrypi/dynamic) where jar are placed

```code
<--Pi4J--> GPIO Blink Example ... started.
Jun 10, 2020 6:43:18 AM com.pi4j.util.NativeLibraryLoader load
SEVERE: Unable to load [libpi4j.so] using path: [/lib/raspberrypi/dynamic/libpi4j.so]
java.lang.UnsatisfiedLinkError: /tmp/libpi4j7035464008728984924.so: libwiringPi.so: cannot open shared object file: No such file or directory
        at java.base/java.lang.ClassLoader$NativeLibrary.load0(Native Method)
        at java.base/java.lang.ClassLoader$NativeLibrary.load(ClassLoader.java:2442)
        at java.base/java.lang.ClassLoader$NativeLibrary.loadLibrary(ClassLoader.java:2498)
        at java.base/java.lang.ClassLoader.loadLibrary0(ClassLoader.java:2694)
        at java.base/java.lang.ClassLoader.loadLibrary(ClassLoader.java:2627)
        at java.base/java.lang.Runtime.load0(Runtime.java:768)
        at java.base/java.lang.System.load(System.java:1834)
        at com.pi4j.util.NativeLibraryLoader.loadLibraryFromClasspath(NativeLibraryLoader.java:159)
        at com.pi4j.util.NativeLibraryLoader.load(NativeLibraryLoader.java:105)
        at com.pi4j.wiringpi.Gpio.<clinit>(Gpio.java:189)
        at com.pi4j.io.gpio.RaspiGpioProvider.<init>(RaspiGpioProvider.java:69)
        at com.pi4j.io.gpio.RaspiGpioProvider.<init>(RaspiGpioProvider.java:51)
        at com.pi4j.platform.Platform.getGpioProvider(Platform.java:125)
        at com.pi4j.platform.Platform.getGpioProvider(Platform.java:118)
        at com.pi4j.io.gpio.GpioFactory.getDefaultProvider(GpioFactory.java:109)
        at com.pi4j.io.gpio.impl.GpioControllerImpl.<init>(GpioControllerImpl.java:53)
        at com.pi4j.io.gpio.GpioFactory.getInstance(GpioFactory.java:91)
        at BlinkGpioExample.main(BlinkGpioExample.java:57)

Exception in thread "main" java.lang.UnsatisfiedLinkError: 'int com.pi4j.wiringpi.Gpio.wiringPiSetup()'
        at com.pi4j.wiringpi.Gpio.wiringPiSetup(Native Method)
        at com.pi4j.io.gpio.RaspiGpioProvider.<init>(RaspiGpioProvider.java:69)
        at com.pi4j.io.gpio.RaspiGpioProvider.<init>(RaspiGpioProvider.java:51)
        at com.pi4j.platform.Platform.getGpioProvider(Platform.java:125)
        at com.pi4j.platform.Platform.getGpioProvider(Platform.java:118)
        at com.pi4j.io.gpio.GpioFactory.getDefaultProvider(GpioFactory.java:109)
        at com.pi4j.io.gpio.impl.GpioControllerImpl.<init>(GpioControllerImpl.java:53)
        at com.pi4j.io.gpio.GpioFactory.getInstance(GpioFactory.java:91)
        at BlinkGpioExample.main(BlinkGpioExample.java:57)
```

savageautomate, 2020-06-18:
Pi4J version 2.0 includes a Java system property you can include to override this default extraction behavior.  Perhaps we should add that support to v1.3 which would at least allow you to bypass the SO extract from JAR logic and just let you tell the library where to find the SO file.

FDelporte, 2020-07-07:
@pavankuppa see https://v2.pi4j.com/library-path for more info

closed 2021-01-11 by savageautomate

savageautomate, 2021-01-11:
Related to #454 

Closing with workaround in Pi4J v1.3.

Added support for system property ("pi4j.library.path") to override default embedded library resolution and allow for user explicitly defined resolution path for `libpi4j.so` native library.

If there is no `/tmp` available, then the Pi4J JAR does not have a location to extract the embedded ("libpi4j.so") native library.  The library will need to be extracted and place on the system and a path provided using this system property.  Note: you can also use "_system_" or "_local_" to have the JAR look for the "libpi4j.so" native library in the system path or in the local runtime directory alongside the JAR file.

commit 6f5b33f3 (2021-01-11): Fixed #491; added support for system property ("pi4j.library.path") to override default embedded library resolution and allow for user explicitly defined resolution path for `libpi4j.so` native library.