Pi4J/pi4j-v1 #509: Pi4J JARs and GPIO Class cannot be used with Java 11

ISO8807, 2020-10-31:
With Java 8 on Windows the following outline code compiles fine against the Pi4J version 1.2 JARs:

  package a.b.c;
  import com.pi4j.wiringpi.Gpio;
  public class D {
    ... Gpio.wiringPiSetupPhys() ...
  }

However, when I try to compile on a Raspberry Pi 4 with Java 11 I get the following error:

  D.java:3: error: package com.pi4j.wiringpi is not visible
  import com.pi4j.wiringpi.Gpio;
	       ^
  (package com.pi4j.wiringpi is declared in the unnamed module, but module com.pi4j.wiringpi does not read it)
  
I am compiling (in outline) with:

`  javac -classpath ...:pi4j-core-1.2.jar:pi4j-device-1.2.jar:pi4j-gpio-extension-1.2.jar D.java`

This is evidently related to the module system introduced in Java 9. It's not obvious whether I can continue to use the Pi4J JARs with Java versions later than 8.

Any ideas? Thanks!

ISO8807, 2020-10-31:
I think I have inadvertently caused this problem. Javac is being invoked from an Ant build script that had the build directory in the classpath as well the Pi4J JARs. This seems to have caused problems because the same Pi4J classes were available in different places. I think it was finding some classes in the build directory and some in the JARs. Having cleaned this situation up, the weird compiler error has disappeared.

closed 2020-10-31 by ISO8807

FDelporte, 2020-11-01:
Thanks for the feedback and solution!

ISO8807, 2020-11-01:
When I hit this strange error, web searches suggested it might need a complicated solution - which is why I posted my query. However, it looks like this was an obscure problem I'd created myself. Maybe someone else can benefit from my experience. Thanks, Ken Turner