Pi4J/pi4j-v1 #511: PI4J with LoRa Module

teceP, 2020-12-08:
Iam trying to send and read data with my LoRa module, which is connected to my raspberry pi 3.

My setup is the following:

[image]

I tried to open on the Default port and on "/dev/ttyS0" but I dont receive any data.

`serial = SerialFactory.createInstance();`
 `serial.open(Serial.DEFAULT_COM_PORT, 115200);`
` //serial.open("/dev/ttyS0", 115200);`
` serial.addListener(getEventListener());`

And my Eventlistener:
`    private SerialDataEventListener getEventListener(){`
     `return serialDataEvent -> {`
            `try {`
               ` console.println("[HEX] " + serialDataEvent.getHexByteString());`
                `console.println("[ASCII DATA] " + serialDataEvent.getAsciiString());`
               ` System.out.println("[Data Received]:[" + Optional.ofNullable(new String(serialDataEvent.getBytes(),` `StandardCharsets.UTF_8)).orElse("Error while getting data.") + "]");`
            `} catch (IOException e) {`
               ` System.out.println("IO EXCEPTION.");`
                `e.printStackTrace();`
            `}`
       ` };`
   ` }`

After I open the port, I try to configure with:
`serial.write("AT+DEST=FFFF");`
`serial.write("AT+RX");`
`serial.write("AT+SAVE");`
`serial.write("AT+CFG=433500000,5,9,6,1,1,0,0,0,0,3000,8,4");`

But I dont receive any data and no one receives any data from me.
What am I doing wrong?

Thanks in advance!

FDelporte, 2020-12-08:
Not sure if this the problem, but are you sure the serial port is not in use by the OS itself?

See https://pi4j.com/1.2/example/serial.html

_Prerequisites

By default, the serial port on the Raspberry Pi is configured as a console port for communicating with the Linux OS shell. If you want to use the serial port in a software program, you must disable the OS from using this port. Please see this blog article by Clayton Smith for step-by-step instructions on how to disable the OS console for this port: http://www.irrational.net/2012/04/19/using-the-raspberry-pis-serial-port/_

savageautomate, 2021-01-11:
**Closed**.  No further activity/response at this time. 
@FDelporte is right, serial port must be **ENABLED** in RPi config and **DISABLED** for system console.

closed 2021-01-11 by savageautomate