Pi4J/pi4j-v1 #512: Not working on Pi Model 4B  V1.2

baptmqt, 2020-12-10:
Pi4J not working on Pi Model 4B  V1.2 
Revision  : c03112 / SoC : BCM2711

```root@rpi:/home/home-control# gpio -v
gpio version: 2.52
Copyright (c) 2012-2018 Gordon Henderson
This is free software with ABSOLUTELY NO WARRANTY.
For details type: gpio -warranty

Raspberry Pi Details:
  Type: Pi 4B, Revision: 02, Memory: 4096MB, Maker: Sony
  * Device tree is enabled.
  *--> Raspberry Pi 4 Model B Rev 1.2
  * This Raspberry Pi supports user-level GPIO access.

root@rpi:/home/home-control# java -jar home-control-1.0-SNAPSHOT-shaded.jar
Unable to determine hardware version. I see: Hardware   : BCM2835
,
 - expecting BCM2708 or BCM2709.
If this is a genuine Raspberry Pi then please report this
to projects@drogon.net. If this is not a Raspberry Pi then you
are on your own as wiringPi is designed to support the
Raspberry Pi ONLY.```

FDelporte, 2020-12-11:
Pi4J V1 uses WiringPi under the hoods, but has been deprecated. Versio 2.52 is the last one and indeed will not support newer versions of the Pi-boards.

Maybe an idea to switch to https://v2.pi4j.com/ ?

eitch, 2020-12-11:
You don't need to switch to pi4j v2, just use the pi4j develop/1.4 branch.
Furthermore there are branches of wiringPi version 1.60 (
https://github.com/eitch/wiringPi/tree/arm64) which work on Pi 4.

fguiet, 2020-12-13:
Hi,

You can use the workraound, I have described here : #510 

Seeya
Fred

savageautomate, 2021-01-14:
That error definitely seems like an outdated version of WiringPi ... or it could be an older version of Pi4J where the WiringPi library was statically compiled.  Either way, I have fully tested the GPIO inputs and outputs on RPi4B for the upcoming Pi4J v1.3 release.  Everything is working properly and the Pi4J v1.3-SNAPSHOT build/artifacts are available now and the full release is coming in the next couple of weeks.  Once caveat however, you must install the _unofficial_ v2.60 of the WiringPi library available here: https://github.com/WiringPi/WiringPi

See:
 - https://pi4j.com/1.3/install.html#WiringPi_Native_Library
 - https://webtechie.be/post/2020-10-10-pi4j-on-raspberry-pi-4/

closed 2021-01-14 by savageautomate