Pi4J/pi4j-v1 #513: PCF8574T INPUT change to LOW and stay

fralu07, 2020-12-18:
Hello everybody,

I am currently trying to connect a keypad to the Raspberry Pi via I2C using PCF8574T.
For this I used the "PCF8574GpioExample". I have adapted this accordingly.

The 3 Input and 4 Output are declared.

The outputs are set to LOW. The inputs are set to High by default. Is it possible to set the inputs to LOW?

A 4x3 keypad is also connected.

Now the problem:

As soon as I press a key on the keypad, the input is set to LOW. But it stays on LOW, even if the key is not pressed anymore.
So I can't press any more keys on the keypad.

I also tried it without the keypad. I closed the input pin to the ground. But also here, the state remains on LOW, if the input is disconnected with Ground.
Tried it with resistors and without. Both did not worked.

Thanks in advance for the support.

Regards,

Francesco

fralu07, 2020-12-21:
More information:
If I set all pins as input, the problem non persist. The signal level return to high, after release the button.
If i set only one pin as output, the signal level on each input pin remain to low, after release the button. The output pin is unused, only provisioned as output. 
I hope this information can help for finding the issue.

savageautomate, 2021-01-15:
I did not see any built-in PULL-UP or PULL-DOWN capability in the code or in the docs.  So this means you would need to have your own PULL-UP or PULL-DOWN resistor in your circuit for each input pin.  Without the pull up/down resistors the input will "**_float_**" and give unpredictable results.  Do you inputs already have pull up/down resistors inline?

closed 2021-02-15 by savageautomate