Pi4J/pi4j-v1 #516: "GpioInterrupt.handleEvents(): Interrupted while waiting for new events!" in Pi4J v1.4

savageautomate, 2021-01-27:
I'm getting "_GpioInterrupt.handleEvents(): Interrupted while waiting for new events!_" printed to the console during a normal Pi4J shutdown in Pi4J v1.4.

REF:
https://github.com/Pi4J/pi4j/blob/develop/1.4/pi4j-core/src/main/java/com/pi4j/wiringpi/GpioInterrupt.java#L211-L229

Testing using `GpioOutputAllExample.java` from a Raspberry Pi 4B running 64-bit Raspbian.

```
pi4j GpioOutputAllExample.java
```

Entered "q" to quit at the program prompt and got the following output:

```
************************************************************
                           GOODBYE                          
************************************************************

GpioInterrupt.handleEvents(): Interrupted while waiting for new events!
```

I think we need to suppress this message either entirely or at least when there is a graceful shutdown/exit.

eitch, 2021-01-27:
ahh, yes. This is of course wrong of me to use an error logger. This is just normal shut down behaviour. Should have just checked if the run flag is false and then end the method, or otherwise continue waiting for events.

savageautomate, 2021-01-27:
Confirmed fix.

closed 2021-01-27 by savageautomate