Pi4J/pi4j-v1 #519: File /lib/raspberrypi/dynamic/libpi4j.so was not found in classpath

ylexus, 2021-02-07:
Using pi4j-core-1.4-20210125.201054-11.jar and getting this:

```
2021-02-06 23:59:09,356 ERROR [main] c.p.u.NativeLibraryLoader Unable to load [libpi4j.so] using path: [/lib/raspberrypi/dynamic/libpi4j.so]
java.io.FileNotFoundException: File /lib/raspberrypi/dynamic/libpi4j.so was not found in classpath.
        at com.pi4j.util.NativeLibraryLoader.loadLibraryFromClasspath(NativeLibraryLoader.java:216) ~[automator.jar:?]
        at com.pi4j.util.NativeLibraryLoader.load(NativeLibraryLoader.java:169) ~[automator.jar:?]
        at com.pi4j.wiringpi.Gpio.<clinit>(Gpio.java:186) ~[automator.jar:?]
        at com.pi4j.io.gpio.RaspiGpioProvider.<init>(RaspiGpioProvider.java:67) ~[automator.jar:?]
        at com.pi4j.io.gpio.RaspiGpioProvider.<init>(RaspiGpioProvider.java:49) ~[automator.jar:?]
        at com.pi4j.platform.Platform.getGpioProvider(Platform.java:117) ~[automator.jar:?]
        at com.pi4j.platform.Platform.getGpioProvider(Platform.java:110) ~[automator.jar:?]
        at com.pi4j.io.gpio.GpioFactory.getDefaultProvider(GpioFactory.java:107) ~[automator.jar:?]
        at com.pi4j.io.gpio.impl.GpioControllerImpl.<init>(GpioControllerImpl.java:51) ~[automator.jar:?]
        at com.pi4j.io.gpio.GpioFactory.getInstance(GpioFactory.java:89) ~[automator.jar:?]
```

I can clearly see Gpio.java tries to load this:
```java
    static {
        NativeLibraryLoader.load("libpi4j.so", "pi4j");
        isrCallbacks = new List[46];
    }
```

which leads to NativeLibraryLoader trying to load `/lib/raspberrypi/dynamic/libpi4j.so` resource from classpath, but only these two are available: under `/lib/raspberrypi/dynamic`: `libpi4j-armhf.so` and `libpi4j-aarch64.so`. This leads to the exception above.

ylexus, 2021-02-07:
Alright, I can see it's a regression that was fixed 10 days ago https://github.com/Pi4J/pi4j/commit/806543d1dc7390f8066433ab78ae8edaa127ef83

closed 2021-02-07 by ylexus

savageautomate, 2021-02-08:
I don't remember deploying updated 1.4-SNAPSHOT -- so maybe that is why it failed for you.  I have deployed new snapshots to the OSS repository based on latest in `master` branch today.