Pi4J/pi4j-v1 #525: Raspberry Pi 4 loses connection when using Pi4J to monitor GPIO inputs

CJure, 2021-02-11:
I have simple GUI application where I want to monitor status of two GPIOs. If I run just GUI without Pi4J code everything works fine, but with GUI code Pi loses network connection over Ethernet port and I have to restart it.
Code:

       ```
 try
        {
            //sound gpio = 23, move = 17
            final GpioController gpio = GpioFactory.getInstance();
            final GpioPinDigitalInput soundSensor = gpio.provisionDigitalInputPin(RaspiPin.GPIO_23, PinPullResistance.PULL_DOWN);
            final GpioPinDigitalInput moveSensor = gpio.provisionDigitalInputPin(RaspiPin.GPIO_17, PinPullResistance.PULL_DOWN);
            moveSensor.setShutdownOptions(true);
            soundSensor.setShutdownOptions(true);
            final Main main = new Main();

            moveSensor.addListener(new GpioPinListenerDigital() {
                @Override
                public void handleGpioPinDigitalStateChangeEvent(GpioPinDigitalStateChangeEvent event) {
                    // display pin state on console
                    System.out.println(" --> GPIO PIN STATE CHANGE: " + event.getPin() + " = " + event.getState());
                    main.jCheckBoxMove.setSelected(event.getState().isHigh());
                }
            });

            soundSensor.addListener(new GpioPinListenerDigital() {
                @Override
                public void handleGpioPinDigitalStateChangeEvent(GpioPinDigitalStateChangeEvent event) {
                    // display pin state on console
                    System.out.println(" --> GPIO PIN STATE CHANGE: " + event.getPin() + " = " + event.getState());
                    main.jCheckBoxSound.setSelected(event.getState().isHigh());
                }
            });
        }
        catch(Exception ex)
        {
            ex.printStackTrace();
            System.out.println("Exception on main: " + ex.toString());
        }
```
If you run similar code in Python I have no problems.

savageautomate, 2021-02-11:
Which pin is GPIO 17?  
[image]
_See:  https://pi4j.com/1.3/pins/rpi-4b.html_

I think pins 17 to 20 were on the earlier models that included the P5 8-pin expansion header.
https://pi4j.com/1.3/pins/rpi-1b-rev2.html

If you meant BCM pin #17, then that would be `GPIO 0` in Pi4J/WiringPi.

[image]

CJure, 2021-02-11:
Thank you for quick response. You are correct I was using BCM pins and it looks that this caused that Ethernet connection would disconnect..

closed 2021-02-11 by CJure