Pi4J/pi4j-v1 #528: java.lang.UnsatisfiedLinkError: /tmp/libpi4j2088695442496528899.so: /lib/arm-linux-gnueabihf/libc.so.6: version `GLIBC_2.28' not found (required by /tmp/libpi4j2088695442496528899.so)

jakusb, 2021-02-21:
feb 21, 2021 2:14:01 AM com.pi4j.util.NativeLibraryLoader load
SEVERE: Unable to load [libpi4j.so] using path: [/lib/raspberrypi/dynamic/libpi4j.so]
java.lang.UnsatisfiedLinkError: /tmp/libpi4j7170120077880840777.so: /lib/arm-linux-gnueabihf/libc.so.6: version `GLIBC_2.28' not found (required by /tmp/libpi4j7170120077880840777.so)
	at java.lang.ClassLoader$NativeLibrary.load(Native Method)
	at java.lang.ClassLoader.loadLibrary0(ClassLoader.java:1938)
	at java.lang.ClassLoader.loadLibrary(ClassLoader.java:1821)
	at java.lang.Runtime.load0(Runtime.java:809)
	at java.lang.System.load(System.java:1086)
	at com.pi4j.util.NativeLibraryLoader.loadLibraryFromClasspath(NativeLibraryLoader.java:227)
	at com.pi4j.util.NativeLibraryLoader.load(NativeLibraryLoader.java:172)
	at com.pi4j.wiringpi.Gpio.<clinit>(Gpio.java:189)
	at com.pi4j.io.gpio.RaspiGpioProvider.<init>(RaspiGpioProvider.java:69)
	at com.pi4j.io.gpio.RaspiGpioProvider.<init>(RaspiGpioProvider.java:51)
	at com.pi4j.platform.Platform.getGpioProvider(Platform.java:125)
	at com.pi4j.platform.Platform.getGpioProvider(Platform.java:118)
	at com.pi4j.io.gpio.GpioFactory.getDefaultProvider(GpioFactory.java:109)
	at com.pi4j.io.gpio.impl.GpioControllerImpl.<init>(GpioControllerImpl.java:53)
	at com.pi4j.io.gpio.GpioFactory.getInstance(GpioFactory.java:91)
	at com.lazymen.pi.gpio.GPIOService.<clinit>(GPIOService.java:11)
	at com.lazymen.portcontroller.LazymenPortController$8.call(LazymenPortController.java:168)
	at com.lazymen.portcontroller.LazymenPortController$8.call(LazymenPortController.java:162)
	at com.lazymen.telegram.Command.execute(Command.java:41)
	at com.lazymen.telegram.LazymenTelegramBot.onUpdateReceived(LazymenTelegramBot.java:481)
	at java.util.ArrayList.forEach(ArrayList.java:1249)
	at org.telegram.telegrambots.meta.generics.LongPollingBot.onUpdatesReceived(LongPollingBot.java:27)
	at org.telegram.telegrambots.updatesreceivers.DefaultBotSession$HandlerThread.run(DefaultBotSession.java:312)

2021-02-21 02:14:01,928 ERROR [LazymenTelegramBot Telegram Executor] telegram.LazymenTelegramBot (LazymenTelegramBot.java:497) - Failed to process incomming update
java.lang.UnsatisfiedLinkError: com.pi4j.wiringpi.Gpio.wiringPiSetup()I
	at com.pi4j.wiringpi.Gpio.wiringPiSetup(Native Method) ~[LazymenPortController-1.3.0-jar-with-dependencies.jar:?]
	at com.pi4j.io.gpio.RaspiGpioProvider.<init>(RaspiGpioProvider.java:69) ~[LazymenPortController-1.3.0-jar-with-dependencies.jar:?]
	at com.pi4j.io.gpio.RaspiGpioProvider.<init>(RaspiGpioProvider.java:51) ~[LazymenPortController-1.3.0-jar-with-dependencies.jar:?]
	at com.pi4j.platform.Platform.getGpioProvider(Platform.java:125) ~[LazymenPortController-1.3.0-jar-with-dependencies.jar:?]
	at com.pi4j.platform.Platform.getGpioProvider(Platform.java:118) ~[LazymenPortController-1.3.0-jar-with-dependencies.jar:?]
	at com.pi4j.io.gpio.GpioFactory.getDefaultProvider(GpioFactory.java:109) ~[LazymenPortController-1.3.0-jar-with-dependencies.jar:?]
	at com.pi4j.io.gpio.impl.GpioControllerImpl.<init>(GpioControllerImpl.java:53) ~[LazymenPortController-1.3.0-jar-with-dependencies.jar:?]
	at com.pi4j.io.gpio.GpioFactory.getInstance(GpioFactory.java:91) ~[LazymenPortController-1.3.0-jar-with-dependencies.jar:?]
	at com.lazymen.pi.gpio.GPIOService.<clinit>(GPIOService.java:11) ~[LazymenPortController-1.3.0-jar-with-dependencies.jar:?]
	at com.lazymen.portcontroller.LazymenPortController$8.call(LazymenPortController.java:168) ~[LazymenPortController-1.3.0-jar-with-dependencies.jar:?]
	at com.lazymen.portcontroller.LazymenPortController$8.call(LazymenPortController.java:162) ~[LazymenPortController-1.3.0-jar-with-dependencies.jar:?]
	at com.lazymen.telegram.Command.execute(Command.java:41) ~[LazymenPortController-1.3.0-jar-with-dependencies.jar:?]
	at com.lazymen.telegram.LazymenTelegramBot.onUpdateReceived(LazymenTelegramBot.java:481) ~[LazymenPortController-1.3.0-jar-with-dependencies.jar:?]
	at java.util.ArrayList.forEach(ArrayList.java:1249) [?:1.8.0_65]
	at org.telegram.telegrambots.meta.generics.LongPollingBot.onUpdatesReceived(LongPollingBot.java:27) [LazymenPortController-1.3.0-jar-with-dependencies.jar:?]
	at org.telegram.telegrambots.updatesreceivers.DefaultBotSession$HandlerThread.run(DefaultBotSession.java:312) [LazymenPortController-1.3.0-jar-with-dependencies.jar:?]

savageautomate, 2021-03-01:
What operating system and version are you running?  RaspberryPi OS Buster? 

Did you try: 
```
sudo apt update 
sudo apt upgrade
```
to get the newest version of the `libc6` package?

jakusb, 2021-03-02:
I am using Raspbian (Linux raspberrypi 4.19.66-v7+)
I resolved the issue bij reverting from version 1.3 to version 1.2.

vishy312, 2023-01-11:
I am getting the same error. I am using the Raspberry Pi Bullseye OS.