Pi4J/pi4j-v1 #532: Unable to open GPIO direction interface for pin [22]

zhouhuabin, 2021-03-22:
I'm testing WiringPiGpioExample.java with JDK8 32bit , use BROADCOM pin schema, every times I got  three errors,broadcom pin number 4 , 20 ,22 export error.  but I write python programs, run it ,all pins are exported is OK. I could not understand why?
**here is output:**
user home is /home/pi/java_code/all_code
CLASSPATH: ../userlib/pi4j-core.jar:../userlib/pi4j-device.jar:../userlib/pi4j-gpio-extension.jar:/home/pi/java_code/all_code:classes ,start path: /home/pi/java_code/all_code 
platform:RASPBERRYPI
<--Pi4J--> GPIO test program
**version:1.3**
export pin=31 is OK.
export pin=30 is OK.
export pin=2 is OK.
export pin=17 is OK.
export pin=16 is OK.
export pin=4 error,msg:Unable to open GPIO direction interface for pin [4]: No such file or directory

start WiringPiGpioExample on pid: 4248 at Mon Mar 22 15:16:25 CST 2021 
export pin=15 is OK.
export pin=3 is OK.
export pin=14 is OK.
export pin=6 is OK.
export pin=13 is OK.
export pin=5 is OK.
export pin=12 is OK.
export pin=8 is OK.
export pin=11 is OK.
export pin=7 is OK.
export pin=10 is OK.
export pin=9 is OK.
export pin=19 is OK.
export pin=18 is OK.
export pin=20 error,msg:Unable to open GPIO direction interface for pin [20]: No such file or directory

export pin=28 is OK.
export pin=27 is OK.
export pin=26 is OK.
export pin=25 is OK.
export pin=24 is OK.
export pin=23 is OK.
export pin=22 error,msg:Unable to open GPIO direction interface for pin [22]: No such file or directory

export pin=21 is OK.
export pin=29 is OK.
java code:
import java.net.URL;
import java.net.URLClassLoader;
import java.util.jar.Attributes;
import java.util.jar.Manifest;
import com.pi4j.io.gpio.GpioFactory;
import com.pi4j.io.gpio.Pin;
import com.pi4j.io.gpio.RaspiBcmPin;
import com.pi4j.io.gpio.RaspiGpioProvider;
import com.pi4j.io.gpio.RaspiPin;
import com.pi4j.io.gpio.RaspiPinNumberingScheme;
import com.pi4j.platform.Platform;
import com.pi4j.wiringpi.Gpio;
import com.pi4j.wiringpi.GpioUtil;

public class WiringPiGpioExample {
    public static void main(String args[]) throws Exception {
        int pin;
        int dataPtr;
        int l, s, d;
        Platform pl = com.pi4j.platform.PlatformManager.getPlatform();
        System.out.println("platform:"+pl.toString());
        
        System.out.println("<--Pi4J--> GPIO test program");
        URLClassLoader cl = (URLClassLoader) (com.pi4j.platform.PlatformManager.class.getClassLoader());
		URL url = cl.findResource("META-INF/MANIFEST.MF");
		Manifest manifest = new Manifest(url.openStream());
		Attributes attributes = manifest.getMainAttributes();
		String version = attributes.getValue("Bundle-Version");//Implementation-
		System.out.println("version:"+version);
        // setup wiringPi
        if (Gpio.wiringPiSetup() == -1) {
            System.out.println(" ==>> GPIO SETUP FAILED");
            return;
        }
        GpioFactory.setDefaultProvider(new RaspiGpioProvider(RaspiPinNumberingScheme.BROADCOM_PIN_NUMBERING));

        Pin[] allPins=RaspiBcmPin.allPins();
        // set all other GPIO as outputs
        for (pin = 0; pin < allPins.length; ++pin) {
            // export all the GPIO pins that we will be using
        	try {
	            GpioUtil.export(allPins[pin].getAddress(), GpioUtil.DIRECTION_OUT);
	            Gpio.pinMode(allPins[pin].getAddress(), Gpio.OUTPUT);
	            GpioUtil.unexport(allPins[pin].getAddress());
                System.out.println("export pin="+allPins[pin].getAddress()+" is OK.");
        	}catch(Exception e) {
        		System.err.println("export pin="+allPins[pin].getAddress()+" error,msg:"+e.getLocalizedMessage());
        	}
        }
    }
}

savageautomate, 2021-03-22:
Which Raspberry Pi model are you using?  If its one of the newer models (4B, 400, CM4, etc) then you will need this version of WiringPi:  https://github.com/WiringPi/WiringPi

There is a section on this page that shows how to install latest WiringPi: 
https://pi4j.com/1.3/dependency.html

zhouhuabin, 2021-03-23:
Thanks for your message. I'm using 4B, wiringPi is 2.52.,and i run command in /sys/class/gpio: echo  3 >export  is OK.
pi@raspberrypi:~ $ gpio -v
gpio version: 2.52
Copyright (c) 2012-2018 Gordon Henderson
This is free software with ABSOLUTELY NO WARRANTY.
For details type: gpio -warranty
Raspberry Pi Details:
  Type: Pi 4B, Revision: 02, Memory: 2048MB, Maker: Sony 
  * Device tree is enabled.
  *--> Raspberry Pi 4 Model B Rev 1.2
  * This Raspberry Pi supports user-level GPIO access.

FDelporte, 2021-03-23:
For full support of Raspberry Pi 4 you will need a newer version of WiringPi.
There is no official one available but Pi4J V1.3 and V1.4 provide a way to get it from a GitHub fork, see
https://pi4j.com/1.4/install.html
`
sudo pi4j --wiringpi`

zhouhuabin, 2021-03-23:
now I had installed the lasted WiringPi to 2.60,but  problems also  exists, why just BCM pin GPIO_04,GPIO_20,GPIO_22 export fail, other bcm pin number export  succeed?
pi@raspberrypi:~ $ gpio -v
**gpio version: 2.60**
Copyright (c) 2012-2018 Gordon Henderson
This is free software with ABSOLUTELY NO WARRANTY.
For details type: gpio -warranty
Raspberry Pi Details:
  Type: Pi 4B, Revision: 02, Memory: 2048MB, Maker: Sony 
  * Device tree is enabled.
  *--> Raspberry Pi 4 Model B Rev 1.2
  * This Raspberry Pi supports user-level GPIO access.
**sudo $JAVA11/bin/java -classpath .:classes:/opt/pi4j/lib/'*' WiringPiGpioExample** 
platform:RASPBERRYPI
<--Pi4J--> GPIO test program
version:1.4
export pin=4 error,msg:Unable to open GPIO direction interface for pin [4]: No such file or directory
export pin=5 is OK.
export pin=20 error,msg:Unable to open GPIO direction interface for pin [20]: No such file or directory
export pin=21 is OK.
export pin=22 error,msg:Unable to open GPIO direction interface for pin [22]: No such file or directory
export pin=23 is OK.
**java code:**
public static void main(String args[]) throws Exception {
        int pin;
        Platform pl = com.pi4j.platform.PlatformManager.getPlatform();
        System.out.println("platform:"+pl.toString());
        
        System.out.println("<--Pi4J--> GPIO test program");
	ClassLoader cl =ClassLoader.getSystemClassLoader();
        JarFile jar = new JarFile("pi4j-core.jar");
	Manifest manifest = jar.getManifest();
	Attributes attributes = manifest.getMainAttributes();
	String version = attributes.getValue("Bundle-Version");//Implementation-
	System.out.println("version:"+version);
        // setup wiringPi
        if (Gpio.wiringPiSetup() == -1) {
            System.out.println(" ==>> GPIO SETUP FAILED");
            return;
        }
        GpioFactory.setDefaultProvider(new RaspiGpioProvider(RaspiPinNumberingScheme.BROADCOM_PIN_NUMBERING));

        Pin[] allPins=new Pin[]{RaspiBcmPin.GPIO_04,RaspiBcmPin.GPIO_05,RaspiBcmPin.GPIO_20,RaspiBcmPin.GPIO_21,RaspiBcmPin.GPIO_22,RaspiBcmPin.GPIO_23};
        // set all other GPIO as outputs
        for (pin = 0; pin < allPins.length; ++pin) {
            // export all the GPIO pins that we will be using
        	try {
	            GpioUtil.export(allPins[pin].getAddress(), GpioUtil.DIRECTION_OUT);
	            Gpio.pinMode(allPins[pin].getAddress(), Gpio.OUTPUT);
	            GpioUtil.unexport(allPins[pin].getAddress());
		    System.out.println("export pin="+allPins[pin].getAddress()+" is OK.");
        	}catch(Exception e) {
			System.err.println("export pin="+allPins[pin].getAddress()+" error,msg:"+e.getLocalizedMessage());
        	}
        }

    }

zhouhuabin, 2021-03-24:
now i run gpio  export command,error is : Device or resource busy, bcm pin number 4,20,22 with eth same error. How could I fix it?
pi@raspberrypi:/sys/class/gpio $ echo 22 > export 
bash: echo: write error: Device or resource busy

zhouhuabin, 2021-06-08:
Finally I resolved my issue , and avoid using  gpio.provisionDigitalInputPin method :
gpio.provisionDigitalInputPin(RaspiBcmPin.GPIO_22, "input",PinPullResistance.PULL_UP);
  I just use Gpio.pinMode(bcm_TiltPin, Gpio.INPUT);
  then read value like this:
       int value = Gpio.digitalRead(bcm_TiltPin);
  not use :
       int value = aihome_tiltPin.getState().getValue();

closed 2021-06-08 by zhouhuabin