Pi4J/pi4j-v1 #533: 1.4 libs: java.lang.NoClassDefFoundError: com/pi4j/component/relay/Relay

kurt-lefevre, 2021-03-22:
I use the Pi4J libraries to control 2 relays on a PiFace Digital board. Up till version 1.3 everything worked fine.
With the 1.3 distro I need following libs: pi4j-core.jar, pi4j-device.jar and pi4j-gpio-extension.jar

I downloaded the 1.4 libs (pi4-distribution) from https://search.maven.org/search?q=com.pi4j and replaced the 1.3 libs with ones in the 1.4 zip.

I use JDK 11.

1.4 gives me following exception at startup:
**Exception in thread "main" java.lang.NoClassDefFoundError: com/pi4j/component/relay/Relay**
at be.relaycontrol.RelayControl.(Unknown Source)
at be.accesscontrol.AccessController.runController(Unknown Source)
at be.accesscontrol.AccessController.main(Unknown Source)
**Caused by: java.lang.ClassNotFoundException: com.pi4j.component.relay.Relay**
at java.base/jdk.internal.loader.BuiltinClassLoader.loadClass(BuiltinClassLoader.java:581)
at java.base/jdk.internal.loader.ClassLoaders$AppClassLoader.loadClass(ClassLoaders.java:178)
at java.base/java.lang.ClassLoader.loadClass(ClassLoader.java:522)

Any hints what I'm doing wrong?

Tx,
Kurt

closed 2021-03-22 by savageautomate

kurt-lefevre, 2021-03-22:
Closed? 
And what's the answer on my question "Do I really need the PiFace Digital specific things or can I just manage the relays via the core API through directly accessing the GPIO pins on the PI board?"

savageautomate, 2021-03-22:
Sorry, I overlooked the question part. My bad ...

> I also had a look at the source code. Do I really need the PiFace Digital specific things or can I just manage the relays via the core API through directly accessing the GPIO pins on the PI board?

So .. **yes** and **no**.  

For any typical basic GPIO exposed directly by the Raspberry Pi 40-pin header that you wish to assert HI or LOW, then yes, you can just use the core GPIO pins.  

However, PiFace provides additional GPIO pins that are controlled thru a separate I/O expander chip.  At a minimum I think you need these classes to create a custom GPIO provider for the PiFace:  
https://github.com/Pi4J/pi4j/tree/release/1.3/pi4j-gpio-extension/src/main/java/com/pi4j/gpio/extension/piface
Using the custom PiFace GPIO controller, then you can just use the Pi4J GPIO APIs exposed by the core library.  You can probably just copy these two classes to your own project sources and not have to include the older Pi4J v1.3 device/extension JARs.

Here is the example code that can provide an example of how to initialize and use the `PiFaceGpioProvider`
https://github.com/Pi4J/pi4j/blob/3fc6d5d8a5fe9501023aa6568a506334b3696466/pi4j-example/src/main/java/PiFaceGpioExample.java

kurt-lefevre, 2021-03-22:
Hi Robert. Confims what I had in mind. Thanks for your answer and the pointers; I'll take it from here.