Pi4J/pi4j-v1 #536: Where is the PCA9685 code?

Sciss, 2021-06-07:
Hi there. I want to use the PCA9685 servo controller board. It seems there was code for this, e.g. https://github.com/Pi4J/pi4j/blob/9a9be997f95798ee6794259c78f8524a64d3c04c/pi4j-example/src/main/java/PCA9685GpioServoExample.java but it has been removed? I also do not see the `PCA9685Pin.java` file. The README ominously states

> Removed pi4j-gpio-extension library. (See comments above)

But there are no "comments above".

Where can I find the classes needed to controls servos via PCA9685, please?

FDelporte, 2021-06-07:
Pi4J is indeed shifting to a "core" approach, providing all generic GPIO functions, but no longer including specific chip or board support. This will make it easier to further maintain and extend the project.

In the CrowPi example project, specific chip support has been worked out as an example alternative approach: https://github.com/Pi4J/pi4j-example-crowpi/tree/b4ad9cdf6607c5ef58068d961067144e945a4c67/src/main/java/com/pi4j/crowpi/components/internal

Sciss, 2021-06-07:
So if I move from 1.3 to 1.4 - because apparently internal sun API is used in 1.3 that doesn't work with OpenJDK 11 - I should be able to compile those extension classes for the PCA9685 from source against 1.4, or has the entire API changed?

Sciss, 2021-06-07:
Yes this approach works (1.4 / OpenJDK 11 / sources of extensions for PCA9685). I get an NPE somewhere in the extensions after clean-up, but set servo positions seems to work.

closed 2021-06-07 by Sciss