Pi4J/pi4j-v1 #542: NoClassDefFoundError com/pi4j/io/gpio/gpiofactory

RobinQuante, 2021-07-20:
Hello,

i have try to use the pi4j lib to controll my outputs from the Raspberry Pi 4.

But i get yt the Start on my RP the Exeption
NoClassDefFoundError com/pi4j/io/gpio/gpiofactory
The log points on the first initialition of the Factory.
i have a ClassNotFoundExeption for the GpioFactory

can someone help me with this problem?

[pom .txt](https://github.com/Pi4J/pi4j/files/6850835/pom.txt)

FDelporte, 2021-07-22:
I see in the pom you are using pi4j version 1.2. For the Raspberry Pi 4, please use version 1.3 or 1.4.

You will also need to update WiringPi to be able to access the GPIOs in the right way.
For full description, see https://pi4j.com/1.4/install.html

`sudo pi4j --wiringpi`

RobinQuante, 2021-07-22:
i have istall a new Linux on the pi,
have install the pi4j with the command from the webside.
i have install the new wiringpi with you Command.
Update the Version of the dependencies in the pom to 1.4
the Native and device i have deleted.

i still have still the same problem.

FDelporte, 2021-07-23:
did you run the program with `sudo`?
if it would be possible to upload your full project to github I can maybe take a look if I can spot the problem...

RobinQuante, 2021-07-24:
i have start the Program with sudo java -jar tractor-1.0-SNAPSHOT.jar

i have try to upload the code and hope you can see it https://github.com/RobinQuante/AutoDriving/tree/main/AutoDrivingTractor

FDelporte, 2021-07-26:
Hey Robin, I've taken a look at your project. Please find attached an updated version.
Nice work BTW with the Swing. If you ever want to move to JavaFX, check https://pi4j.com/getting-started/user-interface-with-javafx/ for more info.

Some of the changes:
* For a correct maven project, the files are moved into a subdirectory
* Package names should be lower case (InOut > inout)
* pom.xml
   * bumped to Java 11 as this is now default on Raspberry Pi OS and for Pi4J V1.4
   * modified the build plugins
* build your project with `mvn package`
* run with `sudo java -jar target/AutoDrivingTractor-1.0-SNAPSHOT-jar-with-dependencies.jar`
* there is still an error with GPIO 17 being used twice, but I leave that one to you to fix ;-)
```
sudo java -jar target/AutoDrivingTractor-1.0-SNAPSHOT-jar-with-dependencies.jar 
Exception in thread "main" com.pi4j.io.gpio.exception.GpioPinExistsException: This GPIO pin already exists: GPIO 17
        at com.pi4j.io.gpio.impl.GpioControllerImpl.provisionPin(GpioControllerImpl.java:554)
        at com.pi4j.io.gpio.impl.GpioControllerImpl.provisionPin(GpioControllerImpl.java:538)
        at com.pi4j.io.gpio.impl.GpioControllerImpl.provisionDigitalInputPin(GpioControllerImpl.java:642)
        at com.pi4j.io.gpio.impl.GpioControllerImpl.provisionDigitalInputPin(GpioControllerImpl.java:670)
        at com.pi4j.io.gpio.impl.GpioControllerImpl.provisionDigitalInputPin(GpioControllerImpl.java:664)
        at com.pi4j.io.gpio.impl.GpioControllerImpl.provisionDigitalInputPin(GpioControllerImpl.java:687)
        at autodrivingtractor.RotationSpeed.<init>(RotationSpeed.java:21)
        at autodrivingtractor.Main.main(Main.java:13)
```

[AutoDrivingTractor.zip](https://github.com/Pi4J/pi4j/files/6877440/AutoDrivingTractor.zip)

closed 2021-07-26 by FDelporte